EmuELEC/EmuELEC #1264: Using sortname as display name is not readable

yingw, 2023-11-27:
in EmuELEC-EmulationStation, when using Detail view and "sort by sortname", the display name in the list will be sortname. I think the real name is better, and sometime the sortname is nonsense to used as display name.

In some non-English area, to sort the game list using their language, they have to use sortname to sort them. But the sortname is some letter (Such as Chinese Pinyin) convert to English letters, it can be sort , but not readable.

Eg:

Using name as sort property, It looks great , but the order is a mess:

[image]

Using sortname as sort prop. Order is good, the looks like this:

[image]

It make the game name unreadable.

So please, can it be reasonable to change this to use name as display always?

BTW, the sortname function is great, works correctly, it helped nonengalish area to sort names.
And, in the Detailed Carousel view, the name is always the name, not sortname, it's great. I think that's the correct solution.

[image]

And the code might be here: [code](https://github.com/EmuELEC/emuelec-emulationstation/blob/232e540e5f56132dbf28e1b0ac31a5682d02bdce/es-app/src/views/gamelist/GameNameFormatter.cpp#L120-L123)

Langerz82, 2023-11-27:
sorting is based on the english alphabet, this is the same with standard emustation.

yingw, 2023-11-28:
I tried to delete that 4 lines code and rebuild image, and then everything runs fine to me .

Langerz82, 2023-12-09:
@yingw I'll add the option to hide sortnames when displaying the gamelists when I get some time soon.

Langerz82, 2023-12-14:
@yingw to test check here:
https://github.com/EmuELEC/emuelec-emulationstation/pull/85

Langerz82, 2024-01-15:
Closing issue as feature to disable show sortname is included in latest revision.

closed 2024-01-15 by Langerz82