EmuELEC/EmuELEC #1277: [Bugs][Bluetooth] Post all Bluetooth Controller Bugs Here!!

Langerz82, 2024-01-18:
Instructions to use new test emuelec-bluetooth file:
https://github.com/EmuELEC/EmuELEC/pull/1276

If it does not work change Debug = True and paste the output in https://gist.github.com then reply to this thread with a link to your debug information.

worstcase-scenario, 2024-01-18:
I never was able to connect my bluetooth controllers (PS5 and a third party PS3 controller) in EmuElec.
So I had to buy two dongles for it to work. Now I just tried  to see if your script changes anything and if I'm able to connect my controllers without the dongles, but it didn't work. Here's the output:

EMUELEC:~ # emuelec-bluetooth
running scan on 60
> bluetoothctl list
bluetooth is already running, exiting.
EMUELEC:~ #

Hope that helps somehow?

Langerz82, 2024-01-18:
@worstcase-scenario that is most likely because the kernel is not recognizing your default bluetooth controller. Not sure why it isn't but may be worth looking into.

psyberpunk, 2024-01-19:
EMUELEC:~ # /emuelec/scripts/emuelec-bluetooth
running scan on 60
> bluetoothctl power on
Changing power on succeeded

> bluetoothctl agent on
> bluetoothctl discoverable on
Changing discoverable on succeeded

> bluetoothctl pairable on
Changing pairable on succeeded

resuming scan
<  Discovery started
<  [CHG] Controller 43:35:B0:07:1F:AC Discovering: yes
[]
<  [NEW] Device 0A:D5:A1:77:63:25 SHANWAN PS3 GamePad
[]
<  [DEL] Device 0A:D5:A1:77:63:25 SHANWAN PS3 GamePad
[]
<  [NEW] Device 0A:D5:A1:77:63:25 0A-D5-A1-77-63-25
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
trusting and pairing device , mac: 0A:D5:A1:77:63:25, icon: input-keyboard, paired: False
> bluetoothctl trust 0A:D5:A1:77:63:25
[CHG] Device 0A:D5:A1:77:63:25 Trusted: yes

Changing 0A:D5:A1:77:63:25 trust succeeded

> bluetoothctl pair 0A:D5:A1:77:63:25
Attempting to pair with 0A:D5:A1:77:63:25

Failed to pair: org.bluez.Error.AuthenticationFailed

successfully trusted  now connect cable.
<  [CHG] Device 0A:D5:A1:77:63:25 Trusted: yes
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 0A:D5:A1:77:63:25 Connected: no
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 0A:D5:A1:77:63:25 Connected: yes
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='', alias='0A-D5-A1-77-63-25', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
resuming scan
<  [CHG] Device 0A:D5:A1:77:63:25 Name: SHANWAN PS3 GamePad
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0), BluetoothDevice(mac='Controller', name='', alias='', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 0A:D5:A1:77:63:25 Alias: SHANWAN PS3 GamePad
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 0A:D5:A1:77:63:25 Modalias: usb:v054Cp0268d0000
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 0A:D5:A1:77:63:25 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  Discovery started
<  [CHG] Controller 43:35:B0:07:1F:AC Discovering: yes
[BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [NEW] Device 4C:89:29:53:0C:39 4C-89-29-53-0C-39
[BluetoothDevice(mac='4C:89:29:53:0C:39', name='', alias='4C-89-29-53-0C-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-101), BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[BluetoothDevice(mac='4C:89:29:53:0C:39', name='', alias='4C-89-29-53-0C-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-101)]
BluetoothDevice(mac='4C:89:29:53:0C:39', name='', alias='4C-89-29-53-0C-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-101)
<  [CHG] Device 0A:D5:A1:77:63:25 ServicesResolved: no
[... cut ...]
[BluetoothDevice(mac='4C:89:29:53:0C:39', name='', alias='4C-89-29-53-0C-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-101), BluetoothDevice(mac='0A:D5:A1:77:63:25', name='SHANWAN PS3 GamePad', alias='SHANWAN PS3 GamePad', clas='0x00000540', icon='input-keyboard', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]

after several trys the controller pair with the bluetooth

Langerz82, 2024-01-19:
@psyberpunk did you connect the controller to the USB cable? I have Shanwan PS3 Gamepad too and it needs a cable connect to authorize the bluetooth to connect, once it tries to pair and fails.

psyberpunk, 2024-01-19:
> @psyberpunk did you connect the controller to the USB cable? I have Shanwan PS3 Gamepad too and it needs a cable connect to authorize the bluetooth to connect, once it tries to pair and fails.

I plug the Shanwan controller with the cable, I used the same controller to move in the emuelec menu and go to controller & bluetooth settings, then to pair bluetooth pad automatically and when the bluetooth scan start I unplug the cable, then i wait for the controller to pair, if doesnt pair I plug again and unplug, almost always after 2 try my controllers pair without any problem, works fine, even after reboot or shutdown, when I start emuelec again just need to press playstation buttons or p3 buttoons and pair without the need to plug cable,

natehsu, 2024-01-19:
Hello, my OS is EmuELEC 4.5, and my controller is an Xbox Wireless Controller Series X. I failed to pair the controller. Here is the log:

<  [NEW] Device 44:16:22:85:8D:A3 Xbox Wireless Controller
[BluetoothDevice(mac='44:16:22:85:8D:A3', name='Xbox Wireless Controller', alias='Xbox Wireless Controller', clas='', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-87), ...
found device Xbox Wireless Controller, mac: 44:16:22:85:8D:A3, icon: input-gaming, paired: False
> bluetoothctl trust 44:16:22:85:8D:A3
[CHG] Device 44:16:22:85:8D:A3 Trusted: yes

Changing 44:16:22:85:8D:A3 trust succeeded

> bluetoothctl pair 44:16:22:85:8D:A3
Attempting to pair with 44:16:22:85:8D:A3

[CHG] Device 44:16:22:85:8D:A3 Connected: yes

Failed to pair: org.bluez.Error.AuthenticationFailed

> bluetoothctl remove 44:16:22:85:8D:A3
[DEL] Device 44:16:22:85:8D:A3 Xbox Wireless Controller

Device has been removed

forgot device Xbox Wireless Controller
<  [CHG] Device 44:16:22:85:8D:A3 Trusted: yes
[]
<  [CHG] Device 44:16:22:85:8D:A3 Connected: yes
[]
<  [CHG] Device 44:16:22:85:8D:A3 Connected: no
[]
<  [DEL] Device 44:16:22:85:8D:A3 Xbox Wireless Controller
[]
<  [NEW] Device 44:16:22:85:8D:A3 Xbox Wireless Controller
[BluetoothDevice(mac='44:16:22:85:8D:A3', name='Xbox Wireless Controller', alias='Xbox Wireless Controller', clas='', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-87), ...]
[BluetoothDevice(mac='44:16:22:85:8D:A3', name='Xbox Wireless Controller', alias='Xbox Wireless Controller', clas='', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-87)]
BluetoothDevice(mac='44:16:22:85:8D:A3', name='Xbox Wireless Controller', alias='Xbox Wireless Controller', clas='', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-87)
found device Xbox Wireless Controller, mac: 44:16:22:85:8D:A3, icon: input-gaming, paired: False
> bluetoothctl trust 44:16:22:85:8D:A3
[CHG] Device 44:16:22:85:8D:A3 Trusted: yes

Changing 44:16:22:85:8D:A3 trust succeeded

> bluetoothctl pair 44:16:22:85:8D:A3
stopping scan
resuming scan

shantigilbert, 2024-01-19:
> > @psyberpunk did you connect the controller to the USB cable? I have Shanwan PS3 Gamepad too and it needs a cable connect to authorize the bluetooth to connect, once it tries to pair and fails.
> 
> @Langerz82 I plug the Shanwan controller with the cable, I used the same controller to move in the emuelec menu and go to controller & bluetooth settings, then to pair bluetooth pad automatically and when the bluetooth scan start I unplug the cable, then i wait for the controller to pair, if doesnt pair I plug again and unplug, almost always after 2 try my controllers pair without any problem, works fine, even after reboot or shutdown, when I start emuelec again just need to press playstation buttons or p3 buttoons and pair without the need to plug cable,

This also needs the patch I just commited to Bluez: https://github.com/EmuELEC/EmuELEC/commit/bfa1b2ceba70a30000bd25b0cf3136eacb6e380d

junm6802030, 2024-01-20:
EMUELEC:~ # emuelec-bluetooth
running scan on 60
resuming scan
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
  File "/emuelec/scripts/emuelec-bluetooth", line 176, in scan_async
    devs = [x for x in bt.devices if x.mac == line.split()[2]]
                       ^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in devices
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in <listcomp>
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 115, in _parse_device_info
    dev.rssi = int(i.split(":")[1].replace("(","").replace(")","").strip())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '0xffffffb6 -74'
resuming scan
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
  File "/emuelec/scripts/emuelec-bluetooth", line 176, in scan_async
    devs = [x for x in bt.devices if x.mac == line.split()[2]]
                       ^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in devices
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in <listcomp>
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 115, in _parse_device_info
    dev.rssi = int(i.split(":")[1].replace("(","").replace(")","").strip())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '0xffffffae -82'
resuming scan
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
  File "/emuelec/scripts/emuelec-bluetooth", line 176, in scan_async
    devs = [x for x in bt.devices if x.mac == line.split()[2]]
                       ^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in devices
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in <listcomp>
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 115, in _parse_device_info
    dev.rssi = int(i.split(":")[1].replace("(","").replace(")","").strip())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '0xffffffa2 -94'
Traceback (most recent call last):
  File "/emuelec/scripts/emuelec-bluetooth", line 331, in <module>
    connected_devs = [dev for dev in bt.devices if dev.connected]
                                     ^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in devices
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 204, in <listcomp>
    return [self._parse_device_info(dev.split()[1]) for dev in devs]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/emuelec/scripts/emuelec-bluetooth", line 115, in _parse_device_info
    dev.rssi = int(i.split(":")[1].replace("(","").replace(")","").strip())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '0xffffffa2 -94'
EMUELEC:~ #

junm6802030, 2024-01-20:
If it does not work change Debug = True

https://gist.github.com/junm6802030/1f882a72bc5bf13aeb358dbae394dee6

junm6802030, 2024-01-20:
The following is the process of manually connecting the Bluetooth controller via SSH. I wonder if there is a conflict caused by two Bluetooths. The built-in Bluetooth in the box is not easy to use, so I connected an external Bluetooth. Can EE specify Bluetooth or disable built-in Bluetooth.

EMUELEC:~ # hciconfig -a
hci1:	Type: Primary  Bus: UART
	BD Address: 43:39:00:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1455 acl:0 sco:0 events:87 errors:0
	TX bytes:4449 acl:0 sco:0 commands:87 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x8f
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'EMUELEC'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.1 (0x7)  Revision: 0x29f
	LMP Version: 4.1 (0x7)  Subversion: 0x6109
	Manufacturer: Broadcom Corporation (15)

hci0:	Type: Primary  Bus: USB
	BD Address: 8C:88:2B:65:1F:2B  ACL MTU: 1021:5  SCO MTU: 255:11
	UP RUNNING PSCAN ISCAN 
	RX bytes:4159 acl:0 sco:0 events:325 errors:0
	TX bytes:29200 acl:0 sco:0 commands:255 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'EMUELEC #1'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.1 (0xa)  Revision: 0xd9a
	LMP Version: 5.1 (0xa)  Subversion: 0x2883
	Manufacturer: Realtek Semiconductor Corporation (93)

EMUELEC:~ # hciconfig hci0 up
EMUELEC:~ # bluetoothctl
Agent registered
[DEL] Device 56:82:D4:2A:56:8D 56-82-D4-2A-56-8D
[DEL] Device 5F:68:7B:9D:C8:1E 5F-68-7B-9D-C8-1E
[bluetooth]# default-agent
Default agent request successful
[DEL] Device 5F:66:B3:27:BC:E5 5F-66-B3-27-BC-E5
[DEL] Device 44:B1:D6:56:EB:92 44-B1-D6-56-EB-92
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 8C:88:2B:65:1F:2B Discovering: yes
[NEW] Device 56:82:D4:2A:56:8D 56-82-D4-2A-56-8D
[NEW] Device 5F:68:7B:9D:C8:1E 5F-68-7B-9D-C8-1E
[NEW] Device 44:B1:D6:56:EB:92 44-B1-D6-56-EB-92
[NEW] Device 5F:66:B3:27:BC:E5 5F-66-B3-27-BC-E5
[NEW] Device B0:10:41:51:95:5C BRAVIA
[NEW] Device C3:C2:08:A7:61:E1 Wireless Controller
[CHG] Device C3:C2:08:A7:61:E1 LegacyPairing: yes
[CHG] Device C3:C2:08:A7:61:E1 LegacyPairing: no
[CHG] Device 5F:68:7B:9D:C8:1E RSSI: 0xffffffae (-82)
[NEW] Device 5A:71:31:33:F8:02 5A-71-31-33-F8-02
[CHG] Device C3:C2:08:A7:61:E1 LegacyPairing: yes
[CHG] Device C3:C2:08:A7:61:E1 LegacyPairing: no
[bluetooth]# trust C3:C2:08:A7:61:E1
[CHG] Device C3:C2:08:A7:61:E1 Trusted: yes
Changing C3:C2:08:A7:61:E1 trust succeeded
[CHG] Device 5A:71:31:33:F8:02 RSSI: 0xffffffba (-70)
[DEL] Device 44:B1:D6:56:EB:92 44-B1-D6-56-EB-92
[bluetooth]# pair C3:C2:08:A7:61:E1
Attempting to pair with C3:C2:08:A7:61:E1
[CHG] Device C3:C2:08:A7:61:E1 Connected: yes
[CHG] Device C3:C2:08:A7:61:E1 Bonded: yes
[CHG] Device C3:C2:08:A7:61:E1 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device C3:C2:08:A7:61:E1 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device C3:C2:08:A7:61:E1 ServicesResolved: yes
[CHG] Device C3:C2:08:A7:61:E1 Paired: yes
Pairing successful
[Wireless Controller]# connect C3:C2:08:A7:61:E1
Attempting to connect to C3:C2:08:A7:61:E1
Connection successful
[Wireless Controller]#

Langerz82, 2024-01-20:
@junm6802030 Try the new script in top post for fix to RSSI Error.
@ebeem might have more knowledge about disabling onboard bluetooth, its over my head.

ebeem, 2024-01-20:
> @junm6802030 Try the new script in top post for fix to RSSI Error. @ebeem might have more knowledge about disabling onboard bluetooth, its over my head.

we usually don't disable a bluetooth controller, instead we configure `bluetoothctl` to use a default one.
I don't know if this will persist between restarts, I assume it should.

This can be done by the executing below terminal commands

first, list your current bluetooth controllers, this should have more than one controller, but to me it's only one
````
~ $ bluetoothctl list
Controller 34:13:E8:C7:95:C9 ebeem [default]
````

second, copy the mac address of the target bluetooth controller and select it as default
```
~ $ bluetoothctl select 34:13:E8:C7:95:C9
````

to confirm everything was properly done, check the output of `bluetoothctl list` to ensure that the target controller has now become the default one.

worstcase-scenario, 2024-01-20:
@Langerz82 
I think I found the reason why my Bluetooth is not working (WiFi works). My android box (A95x_f2) has an MT7668-chip. I searched the net for a possibility to make it work.  And I found this: https://ps4linux.com/mt7668-wifi-bluetooth-drivers-linux-compile/
and this: https://github.com/noob404yt/mt7668-wifi-bt

I am unfortunately not an expert with Linux, but I read all the information and kind of had the feeling that it should somehow be possible to add the driver to the Emuelec kernel if one understands and follows all the instructions. Or am I completely wrong about this?

You guys are definitely the pros and it would be great if you could take a look into this. 

Thanks!

junm6802030, 2024-01-21:
This patch has a low success rate and doesn't work as well for me as last year's patch.

After restarting the box multiple times, it was found that after successful pairing and pairing again after two restarts, a MAC will always be lost.

The log is generated when the two controllers are paired for the first time. At this time, the box is restarted to test whether the pairing is successful again.

https://gist.github.com/junm6802030/c3e7841ba594fd9ef8bc5768023e4a82

junm6802030, 2024-01-21:
As shown in the figure: If there is only one MAC, it will be difficult to pair with the other one.

After the pairing is successful, restart the box. If one of the MACs is lost, the controller cannot be paired at this time. You need to execute scan on to try to pair again.

One of them will be lost every time it is restarted. It is a very strange problem. I thought it was a problem with one of the controllers, but I found that it happened randomly.
[image]

Langerz82, 2024-01-21:
@junm6802030 try this:
https://gist.githubusercontent.com/Langerz82/da02cc12e49c37b5cfe07b3a06b5c15e/raw/4518619fd0a3664ce20a6b187a1fff7b29b27f54/emuelec-bluetooth

junm6802030, 2024-01-22:
> @junm6802030 try this:
> 
> edit slight modification try: https://gist.githubusercontent.com/Langerz82/da02cc12e49c37b5cfe07b3a06b5c15e/raw/f8f731f55dbabf6b30045ce23ceaefcb4a20194e/emuelec-bluetooth
> 
> I will have to check with ebeem that the addition is ok, because ultimately he knows allot more about the script since he coded nearly all of it.
> 
> edit: updated script see above link.

This patch is a great experience for my controller, and pairing can be completed in about three seconds after multiple restarts.

It is recommended to find more Bluetooth controller players to test this patch.

Langerz82, 2024-01-22:
Couple of minor fixes, see:
https://gist.githubusercontent.com/Langerz82/da02cc12e49c37b5cfe07b3a06b5c15e/raw/04a09d8f665cacb478fb260629b089fa00db3f8d/emuelec-bluetooth

junm6802030, 2024-01-22:
The experience of this version is basically the same as the previous version, and the pairing speed is very fast.

This patch is available in my controller pairing.

junm6802030, 2024-01-22:
> The experience of this version is basically the same as the previous version, and the pairing speed is very fast.
> 
> This patch is available in my controller pairing.

The experience of this version is basically the same as the previous version, and the pairing speed is very fast.

This patch is available in my controller pairing.

junm6802030, 2024-01-22:
> Couple of minor fixes, see: https://gist.githubusercontent.com/Langerz82/da02cc12e49c37b5cfe07b3a06b5c15e/raw/04a09d8f665cacb478fb260629b089fa00db3f8d/emuelec-bluetooth

The experience of this version is basically the same as the previous version, and the pairing speed is very fast.

This patch is available in my controller pairing.

shantigilbert, 2024-01-23:
Thank you all for testing!

closed 2024-01-23 by shantigilbert

Langerz82, 2024-01-23:
@shantigilbert Thanks man for the testing and commits.

toughguystudios, 2026-10-02:
Hi, I know this is closed but I'm having a far more frustrating and difficult time than I think was ever intended from the creation of all of this. I'm running a Kinhank Super console X2 that came with EMUELEC 4.6, and I am trying to sync 3 ps4 controllers to the device which come up as "Wireless Controller". To be ultra specific (which I hope you're aware of) One of these ps4 controllers is a version1 ps4 controller circa 2013, the other 2 I believe came out around 2015-2016 which are version2.
I followed your directions of SSHing into my device and using the wget code to install the new script. Currently at this moment I am running the EMUELEC 4.6 with the new script and the 2 version controllers connect, and at best version1 controller connects with them for about a second, and then drops off.

I also tried using the new build 4.8, turning on "Enable Bluetooth Legacy Controllers" and that worked while at home, but when I went to my friends house where I thought we could all play a 5 player game none of the controllers decided to connect at all, all of the suddent, after having religiously tested this 3-4 to solidify my beliefs it was secure.
I went back and forth between ChatGPT and Terminal trying to figure out where the bluetooth issue is. I can provide those logs if need be.

Would you please help me get this thing to the point where the controllers can be consistently connected 100% all the time?
Is this a kinhank device issue? is it an EMUELEC issue? Is it the 2 different types of ps4 controllers? What is the fastest route to rectify this situation? Thank you.

worstcase-scenario, 2026-10-02:
First of all, we do not support any devices that are illegally sold with Emuelec installed, these are modified versions and not the official one, causing much errors. You need to flash a SD card with the official version of Emuelec and check if you still have the BT error. If so, try this: https://github.com/EmuELEC/EmuELEC/pull/1667
You also need an USB cable to connect and pair the controller at the beginning.