EmuELEC/EmuELEC #1313: updatad from 4.6 to 4.7: AUTO LOAD for PSX games not working anymore.

beta-tester, 2024-02-21:
**Describe the bug**
i updated my official EmuELEC v4.6 (installed in december 2023) to v4.7. on v4.6 i had Savestates "AUTO SAVE/LOAD" enabled.
after the update to v4.7, if i start a PSX game, where an auto save state exists, the PSX emulator is loading and closes immediately and jumps back to the EmuELEC game list.
- if i hit "X" on game pad to list save states of the game, and select a save state, the same thing happens. the emulator closes immediately and jumps back to the game list.
- if i hit "X" and select "Start New Game", the emulator opens and i am at the position where the last auto save state left the game - here is expected starting from power on/reset state of a game.
- if i disable "auto save/load" in the "game settings" menu, the emulator opens and starts the game from "power on / reset" state as expected

i also tried to deleted all auto save states, in hope EE v4.7 creates new save states the PSX emulator can load with auto save/load as the version 4.6.
but even the new save states made under v4.7 shows the same strange behavior.

**To Reproduce**
Steps to reproduce the behavior:
1. go to [Start] | Main Menu | Game Settings | Savestates  and enable Auto Save/Load
2. in the PSX game list start a game and play it
3. exit the game as usual to create an auto save state
4. back to the PSX game list, start the same game again
5. the emulator is loading and is closing again - you are back to the PSX game list
6. 
7. with the game selected, hit "X" on the game pad to show the list of save states.
8. start the last save state of the game.
9. the emulator is loading and is closing again - you are back to the PSX game list
   [emuelec (from savestate - load last state).log](https://github.com/EmuELEC/EmuELEC/files/14365090/emuelec.from.savestate.-.load.last.state.log)
10. 
11. with the game selected, hit "X" on the game pad to show the list of save states.
12. from the save state menu, use "Start New Game" instead of an existing state.
13. the emulator is loading and and you are at the same position in the game as the last save state left the game.
[emuelec (from savestate - start new game).log](https://github.com/EmuELEC/EmuELEC/files/14365092/emuelec.from.savestate.-.start.new.game.log)
14. 
15. go to [Start] | Main Menu | Game Settings | Savestates  and disable Auto Save/Load.
16. in the PSX game list start a game and start it as normal
17. the game starts as expected from power on / reset state.

**Expected behavior**
1. if "auto save/load" is enabled i expect to continue the game from the state where i left the game last time.
   but i jumed back to the PSX game list.
2. if i start the last saved state i expect to continue the game from the state where i left the game last time.
   but i jumped back to the PSX game list.
3. if i select "Start New Game" from the savestate menu, i expect to start a new game from power on/reset state.
   but i continued the game at the state where is left the game last time

**EmuELEC version:**
 - official EmuELEC image v4.7 (Amlogic-nr.aarch64)
 
**Device (please complete the following information):**
 - Amlogic S905x2

shantigilbert, 2024-02-21:
can you upload this file? 

`"/storage/playlists/logs/PCSX-ReARMed 32Bit/my favorit private psx game (1).lrtl"`

beta-tester, 2024-02-21:
> can you upload this file?

not much in
```
{
  "version": "1.0",
  "runtime": "1:13:31",
  "last_played": "2024-02-21 22:24:49"
}
```

shantigilbert, 2024-02-22:
Thanks for the report, it seems its a wider issue, not only affecting PSX but others as well

beta-tester, 2024-02-23:
maybe it is "just" an array index +1 offset issue.
if "Last Save State" loads "nothing",
and "Start New Game" loads the "Last Save State".

so everything is shifted by one position.
since the update to 4.7 the key mapping altered for some functions, i don't know how to create a second save state during game play manually.
but if a second save state exists and you select the fist to load, will it start the game from the fist save state or the second save state?

Langerz82, 2024-02-23:
```
Full arguments: /storage/roms/psx/my favorit private psx game (1).chd -Ppsx --core --emulator --controllers=-p1index 0 -p1guid 030003f05e0400008e02000010010000 -wheel  -autosave 1 -state_slot -99
```
The core and emulator arguments are not being supplied, hence why it cannot load anything it seems.

Langerz82, 2024-02-23:
https://github.com/EmuELEC/emuelec-emulationstation/blob/EmuELEC/es-app/src/SaveState.cpp#L75
-1 should be -2 I think so emulator and core arguments get supplied. I will test this out.

Langerz82, 2024-02-23:
For fix and a possible workaround,

Workaround:
I have not tested this but to get it working on the previous code, set the emulator and core in advanced game options and then it may use the config values and run savestates ok.

See:
https://github.com/EmuELEC/emuelec-emulationstation/pull/92/

Langerz82, 2024-02-23:
```
3. if i select "Start New Game" from the savestate menu, i expect to start a new game from power on/reset state.
but i continued the game at the state where is left the game last time
```
I dont think the fix I provided covers this.

beta-tester, 2024-02-23:
at the moment i use:
- AUTO SAVE/LOAD: [ON]
- SHOW SAVESTATE MANAGER: [IF NOT EMPTY]

and then i select [START NEW GAME] to continue the game at last save state.
[START NEW GAME] is already selected so with this procedure i just need to do one button click more than before - it is acceptable as workaround.

if i really want to start a game from power on state, i reset the game.
but normally i want to continue where i left the game.

Langerz82, 2024-02-23:
Give me some time to work this out. All the code is new to me.

miwasp, 2024-02-23:
Can someone please test something?
Create a file ".config/emulationstation/es_savestates.cfg" with the following contents, restart ES and test again:
```
<?xml version="1.0" encoding="UTF-8"?>
<savestates>
  <emulator name="libretro" autosave="true" incremental="true">
    <directory>/storage/roms/savestates/{{system}}</directory>
    <file>{{romfilename}}.state{{slot}}</file>
    <image>{{romfilename}}.state{{slot}}.png</image>
    <autosave_file>{{romfilename}}.state.auto</autosave_file>
    <autosave_image>{{romfilename}}.state.auto.png</autosave_image>
  </emulator>
</savestates>

```

miwasp, 2024-02-23:
Background is that there was a major refactoring of savestates, see here:
https://github.com/EmuELEC/emuelec-emulationstation/commit/4e33901a5bb8d9e979b408d76d155e343a4e8009
There seems to be no fallback to the old way it worked (or at least it does not work in EmuELEC). But there is now a new config file for savestates.

If this works, it is by far not a solution, its just a pointer in the right direction.

shantigilbert, 2024-02-23:
> Background is that there was a major refactoring of savestates, see here: [EmuELEC/emuelec-emulationstation@4e33901](https://github.com/EmuELEC/emuelec-emulationstation/commit/4e33901a5bb8d9e979b408d76d155e343a4e8009) There seems to be no fallback to the old way it worked (or at least it does not work in EmuELEC). But there is now a new config file for savestates.
> 
> If this works, it is by far not a solution, its just a pointer in the right direction.

That is actually a good find, we missed it completely! :/ 

But I actually think this is the solution, unless we add a fallback, or completely revert this change.

shantigilbert, 2024-02-23:
To fully confirm this is the issue, we could change the log to debug, and then look for this:

```
LOG(LogError) << "Could not parse es_savestates.cfg file!";
		LOG(LogError) << res.description();
```

Langerz82, 2024-02-23:
@miwasp I was looking for that file on the net everywhere and couldnt find it. Thanks i'll include it in my testing.

Langerz82, 2024-02-23:
I think I have a fix, although when having auto save/load in ES, when you select start new game, it loads from the auto-save if one is present. It's not a huge deal as you can just reset the game in retroarch menu, but it would be nice for it to work perfectly.

Langerz82, 2024-02-23:
@shantigilbert I can make it so when selecting "Start new game" and "Start new autosave" always load from the start, but I will have to modify ES, and the scripts somewhat. Do you want me to do that, or are you ok with these functions loading the autosave state by default if auto save/load is enabled for save states?

shantigilbert, 2024-02-23:
Well we first have to figure out what is really going on, because if this is a new feature (es_savestates.cfg), I think we should go with it and adopt it as we did with es_features.cfg , as to not have to generate future headaches

Quantum1997, 2024-02-24:
> Can someone please test something? Create a file ".config/emulationstation/es_savestates.cfg" with the following contents, restart ES and test again:
>
> ```
> <?xml version="1.0" encoding="UTF-8"?> [...]

> Can someone please test something? Create a file ".config/emulationstation/es_savestates.cfg" with the following contents, restart ES and test again:
>
> ```
> <?xml version="1.0" encoding="UTF-8"?> [...]

After trying this, for me it opens the game from the EE 4.6 savestates in Game savestate menu (it wasn't opening it before, just like in the case of the person posting this). Savestates from EE 4.6 are set to auto automatically (alltough I previously saved them as slot 0). I can load them if manually loading auto from Retroarch quick menu, but when trying to launch game from Emulation Station from the savestate icon, it launches the game, but just as if I normally launched it, not from the savestate

Also, I still cannot make it for savestates to be shown in EmulationStation  savestates menu (it shows as if I would not have any savestate at all), but they exist and can be normally launched from Retroarch while in game

Quantum1997, 2024-02-24:
> After trying this, for me it opens the game from the EE 4.6 savestates in Game savestate menu (it wasn't opening it before, just like in the case of the person posting this). Savestates from EE 4.6 [...]

Also, forgot to mention that I also tried to create savestate on auto slot, but it still doesn't show it in ES savestates menu, it only shows the savestates that were made in EE 4.6

shantigilbert, 2024-02-24:
The autosaves not loading is another bug that has been identified, it's because retroarch now creates the savestates in folders, ES does not know how to handle that, there is a setting in Retroarch to turn this off, `Sort Save States into Folders by Content Directory` turn it off, then create a savestate, try it, please try it and let us know.

shantigilbert, 2024-02-24:
@Quantum1997 settings should be as follows, with the file provided by @miwasp 
you need to enable advance settings on Retroarch and then go to the Saving menu 
[image]

beta-tester, 2024-02-24:
> @Quantum1997 settings should be as follows, with the file provided by @miwasp you need to enable advance settings on Retroarch and then go to the Saving menu !

@shantigilbert on my settings this option is OFF - is this a must have or a should have ?
with the es_savestates.cfg file, EE4.7 is creating auto save states and can auto load or manually load them from EE UI / SaveState Manager - even this option is not ON.

did you a fresh installation of EE4.7 from scratch or an update from EE4.6 to EE4.7? maybe some of my options are different because of the update.

shantigilbert, 2024-02-24:
Right now, for it to work it needs to be exactly as the screenshot I provided, all settings on and off should be the same, and use the `es_savestate.cfg` file. A proper fix will follow soon, but for now it's just a matter of finding the correct configuration of settings.

Quantum1997, 2024-02-24:
> @Quantum1997 settings should be as follows, with the file provided by @miwasp you need to enable advance settings on Retroarch and then go to the Saving menu ![20240223193845891](https://private-user-images.githubusercontent.com/18634770/307485089-c8ceb245-a3ad-4c0e-ae76-02b56d03c8e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3NTMxMDksIm5iZiI6MTcwODc1MjgwOSwicGF0aCI6Ii8xODYzNDc3MC8zMDc0ODUwODktYzhjZWIyNDUtYTNhZC00YzBlLWFlNzYtMDJiNTZkMDNjOGU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI0VDA1MzMyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5NzZiOWNlNmRkMzBkMGM4MmYzNjJhNjBhMTA4NTFmYjMyNjdjMTA0OGZhMzhkYmFhOTZlMTNmNDg1NzMxNTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZQh9EUQ341LPYSdkyHmGYZ29z4StCA4yGKZMKbWey8Q)

I tried all sorts of combos (starting, of course, with what the screenshot provided). Unfortunately, none of this worked, and the situation is still the same. Fortunately, this doesn't bother me that much, as I used to load savestates from Retroarch anyway, and I guess I can give up the EE 4.6 savestates (didn't have that many of them anyway, the more important were the GBA/PS1 saves, witch work how they should so far). 

The bigger problem is that of the 2nd controller not working from issue #1320 (that issue only talks about PSX, but the problem happens to all other consoles as well, except for the SNES whitch works normally for some reason). I used to play some games with a family member, but now we just can't do this anymore.

shantigilbert, 2024-02-24:
> The bigger problem is that of the 2nd controller not working from issue https://github.com/EmuELEC/EmuELEC/issues/1320

please don't mix issues, keep that discussion on that thread.

Langerz82, 2024-02-24:
Ok so looks like we need es_savestates.cfg + an ES code change to parse the save state filename, and script change to accept the parameter for backwards support for 4.6 and below savestates.

shantigilbert, 2024-02-24:
@Langerz82 @miwasp the solution is even simpler than that, if we don't want to use es_savestates.cfg we just have to change one line 175
https://github.com/EmuELEC/emuelec-emulationstation/blob/a28697fc21b729ddc620f29afead80e5391d67fa/es-app/src/SaveState.cpp#L75

change it to 
`	if (slot >= -1 && this->config != nullptr && !config->isActiveConfig(game) && !racommands)`

This was fixed by f.caruso after I explained the situation, this restores functionality as it was before the change, IF the es_savestates.cfg file does not exist. 

If the `es_savestates.cfg` exists, we need to change a lot more, we need to modify the setsettings.sh file (which I already have) to read the new arguments and create the `.auto` save state to load. 

example provided by f.caruso as well: https://github.com/RetroBat-Official/emulatorlauncher/blob/master/emulatorLauncher/Generators/LibRetro.StateFileManager.cs

advantages of `es_savestates.cfg` is flexibility, as it supports more emulators than just libretro, but we need to determine if it's something we want to use or not.

shantigilbert, 2024-02-24:
btw both solutions need this change in setsettings.sh anyways

```
echo 'savestates_in_content_dir = "false"' >> ${RACONF}
echo 'sort_savefiles_by_content_enable = false"' >> ${RACONF}
echo 'sort_savefiles_enable = "false"' >> ${RACONF}
echo 'sort_savestates_by_content_enable = "false"' >> ${RACONF}
echo 'sort_savestates_enable = "false"' >> ${RACONF}
```

as to keep the savestate manager happy and actually show the states.

closed 2024-02-26 by shantigilbert

merryroc, 2024-03-02:
Thank you shantigilbert! I'll try your solution and test it.

Quantum1997, 2024-03-02:
> btw both solutions need this change in setsettings.sh anyways
>
> ```
> echo 'savestates_in_content_dir = "false"' >> ${RACONF}
> echo 'sort_savefiles_by_content_enable = false"' >> ${RACONF}
> echo [...]

Sorry for the very late answer, been really busy lately! I finally got to try out this sollution, and I can say it works perfectly! Now savestate feature finally works just as intended. Thank you for providing this!

merryroc, 2024-03-03:
Fixed for me, thanks.
with this procedure:
https://github.com/EmuELEC/EmuELEC/issues/1313#issuecomment-1962196701

bitnykk, 2024-06-09:
updating to 4.7 broke several things for me, updates clearly needs WAY more testings !

djibux, 2024-08-22:
> @Quantum1997 settings should be as follows, with the file provided by @miwasp you need to enable advance settings on Retroarch and then go to the Saving menu [...]

I just faced that issue and solved it using the config file shared above. Strangely I must *not* activate the two "folders by content directory" options otherwise save states are stored in a subfolder (`gb/gb` or `psx/psx` instead of simply `gb` or `psx`).

commit a157e879 (2024-02-26): emuelec-emulationstation: Fix savestate manager

Fixes #1313