EmuELEC/EmuELEC #1314: (EmuElec 4.7) RT and LT stop working in the console manufacturers general menu

Quantum1997, 2024-02-22:
**Describe the bug**

The RT and LT buttons of a Xinput controller stop working on the console manufacturers list, alltough they worked just fine in the previous Emuelec version.

**To Reproduce**
Steps to reproduce the behavior:

1. Pressing RT and LT on the XINPUT controller quickly jumps between console manufacturers (example from Nintendo directly to Sega, instead of going through all Nintendo consoles one by one).
2. Going into Controller key redefine (in Emuelec controller submenu), to manually set the keys for the controller, as well as the hotkey
3. After setting the keys is complete, RT and LT buttons won't quickly skip between consoles anymore

**Expected behavior**
RT and LT trigger buttons of the XINPUT controller should skip between console manufacturers without having to browser through every single console of the respective manugacturer.

**Screenshots**
No need to.

**EmuELEC version:**
 - Version (4.7)

**Device (please complete the following information):**
 - SOC: [S095x3]
 - Brand: [HK1 BOX]
 - Model [Amlogic S905X3]
 - Version [Android 9]

**Additional context**
I also noticed that, while being in a Retroarch emulator, while previously, in 4,6, I was able to access retroarch menu by also pressing R thumb + L thumb buttons, this time I cannot do so anymore in 4.7. The difference here is that I am not sure if the key redefine or something else is affecting this.

worstcase-scenario, 2024-02-26:
To access the retroarch menu, I have found out that you have to press Hotkey plus Triangle.

shantigilbert, 2024-02-26:
It seems Retroarch changed how this works, but I have not investigated as this has not been chaged by EmuELEC

Quantum1997, 2024-03-03:
> It seems Retroarch changed how this works, but I have not investigated as this has not been chaged by EmuELEC

Well, tbh, for me this is not that much of a problem anyway. This is certainly not a big problem,because we can, as alternative, use left/right arrows of gamepad to navigate just as we would do with RT/LT in the console menu. And in Retroarch we can still HOTKEY + TRIANGLE/Y to go into the menu. It's not as easier as it was before, but it's still more than manageable (at least for me).

Quantum1997, 2024-03-03:
But at the same time I felt like, since I went on Github, I should let you know of this problem as well, just in case :)

Langerz82, 2024-03-10:
Since L3+R3 is not working in retroarch, perhaps we change it to a default of SELECT hold 2 seconds?

Kissjozsef3, 2024-03-13:
I have set up Emuelec 4.7 yesterday and also experienced the same issue with LT and RT buttons.
I tried to play Tomb Raider,where the mentioned buttons are set up,that Lara steps left and right,but unfortunately nothing happens,although I have adressed these 2 buttons in the controller mapping in Emuelec...
Otherwise everything else is ok and fantastic with the new Emuelec.

Langerz82, 2024-03-30:
Possibly related to RA bug:
https://github.com/EmuELEC/EmuELEC/issues/1343#issuecomment-2027924040

Langerz82, 2024-03-31:
Pretty sure this is fixed in the latest RA bump. If you do the latest dev build and still have this issue let me know.

closed 2024-03-31 by Langerz82