EmuELEC/EmuELEC #1316: GBA/GBC games require settings real-time clocks.

Vodovoz58, 2024-02-23:
Such games how Pokemon series, Harvest Moon (GBC) and maybe others require a system time change.
Could you please add such a setting to the interface EmuELEC?

shantigilbert, 2024-02-24:
You need to elaborate how this is enabled

Vodovoz58, 2024-02-24:
Add a time/date setting to the menu Emuelec. Like in Android. 
You can exit the emulator there. Change the time. And get back into the game.

shantigilbert, 2024-02-26:
How to enable in EmuELEC not Android.

Vodovoz58, 2024-02-27:
I don't understand your question. These games cannot be played in EmuELEC. To upgrade tools in Harvest Moon I have to wait for 3 hours of real time. If EmuELEC had a time/date setting I would have solved this issue in 2 minutes.

worstcase-scenario, 2024-02-27:
You can just change the time zone in the system settings (from EUROPE to PACIFIC, for example) and the system-time will automatically be changed.

Or you can ssh into Emuelec (or use the terminal in setup) and change the date via the "date -s hh:ss" command