EmuELEC/EmuELEC #1318: Sonic 1 & 2 crashing when trying to start game - no info in emuelec logs on the crash

DingleFlute, 2024-02-23:
**Describe the bug**
Sonic 1 & 2 crashing when trying to start game - no info in emuelec logs on the crash

**To Reproduce**
Steps to reproduce the behavior:
Start Sonic 1 or Sonic 2 port
Select Start Game
Crashes back to UI

**EmuELEC version:**
Version 4.7

**Device (please complete the following information):**
SOC: S922 beelink gt-king

**Additional context**
Looks like we have version 1.1.2 and from having a quick look at https://github.com/Rubberduckycooly/Sonic-1-2-2013-Decompilation/releases. 1.1.2 had a critical bug  (not sure if it is the first release or the fixed release that is getting installed but with the issue i am having it could be the broken one). maybe we could get a bump to a newer version? or point me in the right direction and i can update the version myself on the package (i haven't been able to figure out how to view packages that aren't viewable "publicly").

miwasp, 2024-02-23:
Works for me.
Are you sure it crashes? Maybe you just exit the game?
Do you select "Start Game" with Button A, B or Start?

DingleFlute, 2024-02-23:
sorry, it looked a lot like a crash but i was unintentionally exiting.... thanks for setting me straight

closed 2024-02-23 by DingleFlute