EmuELEC/EmuELEC #1320: [update from EE4.6 to 4.7]: in PSX, the second joystick isn't working anymore.

beta-tester, 2024-02-23:
**Describe the bug**
i updated my official stable EE4.6 to EE4.7.
after the update my second joystick isn't working anymore in PSX games.
with EE4.6 i was able to play a two player two joystick game without issues.
now the second joystick isn't doing anything, only holding the [MENU] button is still working on the second joystick, to open the [Quick Menu] of the emulator.

**To Reproduce**
Steps to reproduce the behavior:
1. start EE4.7
2. plug in a second joystick
3. second joystick is detected by EE4.7 and you get the message that the second joystick is connected
4. open a PSX two player two joystick game (e.g.: CTR - Crash Team Racing)
5. at start of the game, EE shows that the first joystick is configured to port 1 and the second is configured to port 2.
6. start a two player two joystick game session
7. the first player is able to move the game character as usual
8. the second player can't do anything - the character does not move or react at all.

**Expected behavior**
i expect in PSX that with two connected joysticks and two from EE marked as configured joysticks will work well as it was woking under EE4.6.

**EmuELEC version:**
 - Version 4.7

**Device (please complete the following information):**
 - SOC: Amlogic S905X2 (2GB RAM, 100Mb/s)

**Additional context**
- i tried to swap the two cheap chinies joysticks with two genuim XBOX 360 joysticks but the behavio is the same, the second player isn't able to do anything in the game with the second joystick.
- i compared with my older backup of my EE4.6 SD card with the updated EE4.7, but i can't find differences in joystick settings that would explain the different behavior.

beta-tester, 2024-02-24:
i have to correct my previous EDIT a bit, that holding the [MENU] button for a long time will open the [Quick Menu].
holding the [MENU] button for a long time (ca. >5s) will swap the joystick with a "Shanwan SKY Andriod X2" one (what ever that is) by disconnecting the normal used joystick. the disconnect lets pause the game and the [Quick Menu] opens to wait for resuming the game.

at the moment, holding the [MENU] button to swap the joysticks is the only known way for me to get the [Quick Menu] indirectly. with EE4.6 i could it directly open by pressing left and right analog middle buttons. but after the update to EE4.7 these button combo does not do anything and i haven'd figured out how do open the [Quick Menu] directly under EE4.7 the normal way.
is this another bug/issue to report?

beta-tester, 2024-02-24:
i tried to use a 4 player 4 joystick game with 4 joysticks, without success. :-(
same behavios. only the 1'st joystick is working.

**To Reproduce**
1. connect all 4 joysticks to EE4.7
   e.g.:
   2x cheap wireless PS1 clones, shown up as "Microsoft X-Box 360 pad" in EE,
   2x genuine Microsoft XBox 360 Wireless Controller, shown up as "Xbox 360 Wireless Receiver"
2. start a 4 player 4 joystick game (e.g.: "Crash Bash" for PSX)
3. enter the [Quick Menu]
4. go to "Core Options"
5. go to "Input"
6. go to "Multitap Mode"
7. select "Port 1"
8. go 1 step back to "Core Options"
9. go to "Manage Core Options"
10. select "Save Game Options"
11. go 4 steps back to "Main Menu"
12. select "Settings"
13. go to Input"
14. go to "RetroPad Binds"
15. go "Port 1 Controls"
15.1. check/correct "Device Index" to your 1'st joystick
15.2. check correct all other settings of the 1'st joystick settings
15.3. in case you needed to correct settings, select "Save Controller Profile"
15.4. go 1 step back to "RetroPad Binds"
16. go "Port 2 Controls"
16.1. check/correct "Device Index" to your 2'nd joystick
16.2. check/correct all other settings of the 2'nd joystick settings
16.3. in case you needed to correct settings, select "Save Controller Profile"
16.4. go 1 step back to "RetroPad Binds"
17. go "Port 3 Controls"
17.1. check/correct "Device Index" to your 3'rd joystick
17.2. check/correct all other settings of the 3'rd joystick settings
17.3. in case you needed to correct settings, select "Save Controller Profile"
18.4. go 1 step back to "RetroPad Binds"
18. go "Port 4 Controls"
18.1. check/correct "Device Index" to your 4'th joystick
18.2. check/correct all other settings of the 4'th joystick settings
18.3. in case you needed to correct settings, select "Save Controller Profile"
19. go 3 steps back to "Settings"
20. select "Main Mernu"
21. go to "Configuration File"
22. select "Save Current Configuration"
23. go 1 step back to "Main Menu"
24. go to "Quick Menu"
25. select "Close Content"
26. you are back to "Main Menu"
27. select "Quit"
28. you are back in the Game List.
29. open the "SaveState Manage" of the selected game (e.g.: "Crash Bash" for PSX)
30. delete the last Auto Save State if existing
31. select "Start New Game"
32. while the game is starting, all 4 joysticks shown up as "... Configured tp Port ..."
33. in the game, if you select a battel for 4 players, only the 1'st player gets control.
the others can't do anything - no control to the characters.

shantigilbert, 2024-02-24:
Try setting the buttons directly in the retroarch input settings (global) for the extra controllers, I just tested this with 2 controllers and it worked for me. If I didn't do that then the 2nd player could not be controlled. This seems to be an issue with retroarch, either something changed, or its broken.

beta-tester, 2024-02-24:
i got it working!

** reset all settings to defaults **
1. in EE "main menu"
1.1. "emuelec settings"
1.2. "danger zone"
1.3. "reset retroarch config to default"
1.4. "reset system to default config"

** setup global EE settings **
```
1. in EE "mein menu"
1.1. "controller & bluetooth settings"
1.2. "controller mapping"
   - do the mapping for joystick 1
   - do the mapping for joystick 2
   - do the mapping for joystick 3
   - do the mapping for joystick 4

2. start a PSX game and enter RA "quick menu".
    only one joystick is working in PSX and RA
2.1. "close content"
2.2. in RA "main menu", select "settings" > "input" > "retropad binds"
2.3. "port 1 controls"
   - setup all controlls (even you see assignments exists)
   - "save controller profile"
   - go one step back to "input"
2.4. "port 2 controls"
   - setup all controlls (even you see assignments exists)
   - "save controller profile"
   - go one step back to "input"
2.5. "port 3 controls"
   - setup all controlls (even you see assignments exists)
   - "save controller profile"
2.6. "port 4 controls"
   - setup all controlls (even you see assignments exists)
   - "save controller profile"
   - go one step back to "input"
2.7. from now on all four joysticks are working in RA but not in PSX yet.
   - go back to "main menu"and "quit"

3. in EE start a 4 player 4 joystick PSX game and enter the RA "quick menu"
3.1. "core options" > "input", set "multitap mode" to "port 1"
3.2. go one step back to "core options", and to "manage core options"
    - "save game options" (thi is per game)
    - (the option "save content directory options"i for all PSX games but can "break" 1  or 2 player games)
3.3. go back to "quick menu" and select "cloce content" and "quit"
3.4. from now on all PSX games can be played by 4 players with 4 joysticks
     tested with "Crash Bash", "CTR - Crash Team Racing", "Team Buddies", "Pong", "Break Out", ("Frontschweine").
```

i think from the update process from EE4.6 to EE4.7 some old configurations broke the 4 player settings in EE4.7.

closed 2024-02-24 by beta-tester

Quantum1997, 2024-02-24:
> Try setting the buttons directly in the retroarch input settings (global) for the extra controllers, I just tested this with 2 controllers and it worked for me. If I didn't do that then the 2nd [...]

It finally worked. Alltough keys were displaying as allready being set in RA global input settings, I have done as you said, resetting them again, as if I would've done it in ES controller setup. After doing so, the 2nd controller now work. Tested on a few games in NES and GENESIS and 2nd controller is now in control as well (even after changing controls in quick menu input and saving core remap files). Thank you for providing the solution, it's mindblowing to know it was that simple to fix