EmuELEC/EmuELEC #1323: Impossible to complete 8bitdo M30 2.4G (White color) PAD key mapping correctly

nevmind, 2024-02-25:
Describe the bug
8bitdo M30 2.4G(White color version) pad is connected with ODROID N2 PLUS
(https://www.8bitdo.com/m30-2-4g-for-genesis-mini-mega-drive-mini/)
But Key mapping doesn't work correctly so couldn't finish mapping successfully.

1. D-PAD LEFT is not recognized when I press D-PAD LEFT once,
but is recognized when I press D-PAD LEFT once more
2. D-PAD RIGHT doesn't work at all. so impossible to map D-PAD RIGHT

To Reproduce

plug in 2.4G receiver of 8bitdo M30 
check connection is working or not
Expected behavior
Successful connection & Key mapping

Screenshots
[image]

EmuELEC version:

Version 4.7
Device (please complete the following information):

SOC: [Amlogic S922X]
Brand: [Hardkernel]
Model [Odroid N2 PLUS]
Version [N/A]
Additional context
N/A

nevmind, 2024-02-25:
@shantigilbert 
8bitdo Ultimate 2.4G controller works good now. But there is still problem with 8bitdo M30 2.4G controller mapping.

tvboxprivada, 2024-02-25:
> Describe the bug 8bitdo M30 2.4G(White color version) pad is connected with ODROID N2 PLUS as "6B controller" (https://www.8bitdo.com/m30-2-4g-for-genesis-mini-mega-drive-mini/) But Key mapping doesn't work correctly so couldn't finish mapping successfully.
> 
> 1. D-PAD LEFT is not recognized when I press D-PAD LEFT once,
>    but is recognized when I press D-PAD LEFT once more
> 2. D-PAD RIGHT doesn't work at all. so impossible to map D-PAD RIGHT
> 
> To Reproduce
> 
> plug in 2.4G receiver of 8bitdo M30 and try to map each key Expected behavior Successful Key mapping
> 
> Screenshots ![image](https://private-user-images.githubusercontent.com/5904762/307573401-d02d13b7-c8a1-40c6-b318-7ca996609d8d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg4NzQ2MjQsIm5iZiI6MTcwODg3NDMyNCwicGF0aCI6Ii81OTA0NzYyLzMwNzU3MzQwMS1kMDJkMTNiNy1jOGExLTQwYzYtYjMxOC03Y2E5OTY2MDlkOGQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjVUMTUxODQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWZiZjRiNzRiMTM4NWQzZWFlNDU0YWUwY2VlYWQzZjMwOWI0YzMxYmQzMTY2ZWQwM2MwZmJhNmFmM2M2ZjQ5NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.bJqpujBu1P9rRh9kE6C9-Y9aRXm6gMHZ9Z7GP7W4gXU)
> 
> EmuELEC version:
> 
> Version 4.7 Device (please complete the following information):
> 
> SOC: [Amlogic S922X] Brand: [Hardkernel] Model [Odroid N2 PLUS] Version [N/A] Additional context N/A

I have an Amlogic 905x3, and I'm having exactly the same problem with my 8bitdo M30 2.4G gamepad

shantigilbert, 2024-02-25:
closed as its related to #1245 1245

shantigilbert, 2024-02-26:
@nevmind the pad is being detected as a Sega Genesis pad (I think), it never actually had any analog sticks, on the screenshot you provided it is detecting the pad as an axis (analog) so something is confusing it, but like I said on the other issue, I do not know how to fix that, I will reopen this issue to see if someone else has any other ideas.

meanwhile you might want to provide additional information:
like: 
`lsusb`
complete `dmesg` (use a paste service like pastebin)

nevmind, 2024-02-27:
Check my result please.

after connecting 8bitdo ultimate 2.4G
[ 25.136849@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3106
after connecting 8bitdo M30 2.4G
[ 281.370491@2]- usb 1-1.4: New USB device found, idVendor=0ca3, idProduct=0024

and so I tried following

copy /usr/lib/udev/rules.d/99-8bitdo-bluetooth-controllers.rules to /storage/.config/udev.rules.d
modify like this
#8Bitdo M30
ACTION=="add", ATTRS{idVendor}=="0ca3", ATTRS{idProduct}=="0024", RUN+="/sbin/modprobe xpad", RUN+="/bin/bash -c 'echo 0ca3 0024 > /sys/bus/usb/drivers/xpad/new_id'"
Reboot and connected as "6B controller"
try mapping again.
But exactly same bug is reproduced.

nevmind, 2024-02-29:
@shantigilbert 
More information.
D-pad Left&Right is working well in retroarch pad config and real game.
Could be hint to debug? D-pad mapping problem is reproduced only in emulstation key mapping.

D-pad Left : Axis-0, 
D-pad Right : Axis+0
[image]

nevmind, 2024-02-29:
@shantigilbert 
This is resolved. There is mode change method in M30 2.4G pad.(I couldn't find this in manual or 8bitdo homepage)

Home+up (5sec) : x-input mode
Home+left (5sec) : switch mode
Home+down (5sec) : d-input mode

1. change x-input mode 
2. then M30 is connected with "X-box 360 pad"
3. keymap in emulstation
4. D-pad is now recognized as "Hat0 up/down/left/right" correctly.

Thanks.

nevmind, 2024-02-29:
@tvboxprivada 
Check my resolution

shantigilbert, 2024-02-29:
> @shantigilbert This is resolved. There is mode change method in M30 2.4G pad.(I couldn't find this in manual or 8bitdo homepage)
> 
> Home+up (5sec) : x-input mode Home+left (5sec) : switch mode Home+down (5sec) : d-input mode
> 
> Select+up (5sec) : d-pad only mode Select+left (5sec) : dpad as left analog Select+right (5sec) : dpad as right analog Select+down (5sec) : swap A/B, X/Y
> 
> 1. change x-input mode
> 2. then M30 is connected with "X-box 360 pad"
> 3. change dpad only mode
> 4. keymap in emulstation
> 5. D-pad is now recognized as "Hat0 up/down/left/right" correctly.
> 
> Thanks.

Oh, yeah 8bitdo has different modes. 
After doing this did you had to add or edit any udev.rules?

nevmind, 2024-03-01:
No need to add or edit udev.rules

tvboxprivada, 2024-03-01:
Have you found a solution to this problem?

shantigilbert, 2024-03-01:
Then I will consider this issue complete, thank you!

closed 2024-03-01 by shantigilbert

tvboxprivada, 2024-03-03:
> Oh, yeah 8bitdo has different modes. After doing this did you had to add or edit any udev.rules?

8bitdo possui dois tipos de gamepads:
1) Modelo Bluetooth
2) Modelo Wirelles 2.4Ghz

Essa opção de solução será possível somente no gamepad Modelo Bluetooth

Sendo assim o problema em questão permanece no modelo wirelles 2.4Ghz

Segue link do video, apenas como apresentação, da existem dos dois tipos de gamepads diferentes

https://www.youtube.com/watch?v=4Dzyak3dnXg

tvboxprivada, 2024-03-03:
> Oh, yeah 8bitdo has different modes. After doing this did you had to add or edit any udev.rules?

​
324 / 5.000
Resultados de tradução
Resultado da tradução
8bitdo has two types of gamepads:
1) Bluetooth Model
2) Wirelles 2.4Ghz model

This solution option will only be possible on the Bluetooth model gamepad

Therefore, the problem in question remains in the 2.4Ghz wireless model

Follow the video link, just as a presentation, so there are two different types of gamepads

https://www.youtube.com/watch?v=4Dzyak3dnXg

tvboxprivada, 2024-03-03:
Additional information follows

tvboxprivada, 2024-03-03:
Additional information follows introducing the two gamepads

1) 8Bitdo M30 Bluetooth Model: 
https://www.8bitdo.com/m30/ 
(problem solved)

2) 8Bitdo M30 Wirelles 2.4Ghz model: 
https://www.8bitdo.com/m30-2-4g-for-genesis-mini-mega-drive-mini/
(incompatibility problem remains)

tvboxprivada, 2024-03-03:
Additional information follows introducing the two gamepads

8Bitdo M30 Bluetooth Model:
https://www.8bitdo.com/m30/
(problem solved)
[image]

8Bitdo M30 Wirelles 2.4Ghz model:
https://www.8bitdo.com/m30-2-4g-for-genesis-mini-mega-drive-mini/
(incompatibility problem remains)
[image]