EmuELEC/EmuELEC #1324: Problem with 8bitdo SN30 Pro and 8bitdo Pro 2 controllers in Emuelec 4.7

alonsocr23, 2024-02-28:
**Describe the bug**
I have tested 2 8bitdo bluetooth controllers in emuelec 4.7, specifically the 8bitdo SN30 Pro and 8Bitdo Pro 2 models. The problem is when entering a game (in retroarch) the 8bitdo controllers do not respond and the only solution I have found It is also configured within retroarch and that is how it is solved, but it is not how it should be. For example, in emuelec 4.5 this problem did not happen to me.

**EmuELEC version:**
 - Version 4.7

**Device (please complete the following information):**
 - SOC: S905x3
 - Brand: HK1Box
 - Model: 4gb RAM
 - Version: Android 9

shantigilbert, 2024-03-03:
Please reconfigure your controllers in ES

alonsocr23, 2024-03-09:
> Please reconfigure your controllers in ES

You're right, I tried it again and it works, I did something wrong. Thank you for your great work shantigilbert. You can now delete this report

closed 2024-03-10 by Langerz82