EmuELEC/EmuELEC #1338: PCE-CD achievements not working / crashing

MagicD3VIL, 2024-03-11:
**Describe the bug**
When I launch any Turbografx 16 game supporting RetroAchievements, it works correctly. However, any Turbografx 16 CD game supporting achievements I launch (tg16cd) for some reason starts with the Achievements option disabled in Retroarch. If I manually enable the Achievements option in the Retroarch main menu and then navigate to the quick menu to open the achievements tab, then the emulator crashes with 
```bash
/usr/bin/emuelecRunEmu.sh: line 472: 5068 Segmentation fault ${RABIN} ${VERBOSE} -L /tmp/cores/${EMU}.so --config ${RACONF} "${ROMNAME}" exit 139
```

It also seems that the emulator is not able to calculate the rom hash of any tg16cd game. Going to the Information tab in Quick Menu after enabling achievements always shows the RetroAchievements Hash as N/A.

Both tg16 and tg16cd are using the Beetle PCE FAST core.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch tg16cd game
2. Open Retroarch overlay
3. Navigate to the Main Menu
4. Settings > Achievements > ON
5. Main Menu > Quick Menu > Achievements
6. The red EmuELEC Run Log appears as the emulator crashed

**Expected behavior**
The achievements should work in the tg16cd games as they do in the tg16 games. Opening the achievements quick menu after enabling achievements should not crash the emulator.

**EmuELEC version:**
 - Version 4.7

**Device (please complete the following information):**
 - SOC: Amlogic S905X rev d

**Additional context**
I am not sure if this issue should be opened on the libretro/RetroArch repo or here. Please correct me if this issue cannot be resolved from the EmuELEC side and should be opened elsewhere.

Cheers

shantigilbert, 2024-03-12:
post the complete log

MagicD3VIL, 2024-03-12:
Here is the full [emuelec.log](https://github.com/EmuELEC/EmuELEC/files/14575812/emuelec.log), as requested.

For a test, I tried launching the same ROM on my other Android aarch64 device that is running pure RetroArch (also v1.17.0), and there the achievements worked correctly. The only difference I noticed in the configuration is that EmuELEC's Beetle PCE Fast is `v1.31.0.0 e8dc7fe9fa14` while the pure RA install on my other device uses Beetle PCE Fast `v1.31.0.0 f450a71`.

Langerz82, 2024-07-24:
@MagicD3VIL 
Can you update to EmuELEC 4.8 TEST and let me know if there is still an issue?
https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024

You can do it from emulationstation in the update TEST channel.

Langerz82, 2024-09-17:
@MagicD3VIL did updating to 4.8TEST fix the issue?

MagicD3VIL, 2024-09-17:
@Langerz82 Emulationstation tells me that no updates are available in the Beta channel (I don't have the TEST channel available) so I'll guess I'll have to update manually.

MagicD3VIL, 2024-09-17:
@Langerz82 It now crashes when launching the game on
`[ERROR] [Core]: Frontend is built for dynamic libretro cores, but path is not set. Cannot continue`

With warning before that saying that
`[WARN] --libretro argument "/tmp/cores/mednafen_pce_fast_libretro.so" is not a file, core name or directory. Ignoring.`

Langerz82, 2024-09-17:
Sounds like your install might be corrupt. Try using a different sd card and re-download the image file.

MagicD3VIL, 2024-09-17:
I tried updating with the Generic-img.gz first, but since that didn't do anything I deleted it and downloaded the .tar as per instructions here https://github.com/EmuELEC/EmuELEC/wiki/How-to-update , since img.gz is not mentioned in the instructions at all. Was I supposed to do something with the img file too?

Langerz82, 2024-09-17:
This is the file you should put in your update path:
[EmuELEC-Amlogic-ng.aarch64-4.8-TEST-04212024.tar](https://github.com/EmuELEC/EmuELEC-tests/releases/download/v4.8-TEST-04212024/EmuELEC-Amlogic-ng.aarch64-4.8-TEST-04212024.tar)

I think you are better off, rolling back on your current version. And installing 4.8TEST on a clean card as an image.

Langerz82, 2024-09-17:
I tested the same emulator with retroachievements and it worked for me, so I'm pretty sure it's fixed. Not sure if out 4.8TEST version has been updated or not. I could access the game achievements via the quick menu, and resume play.

Langerz82, 2024-09-20:
I'm going to close this issue as it should be fixed in the latest Dev version. If it isnt feel free to reply and I'll re-open the issue.

closed 2024-09-20 by Langerz82