EmuELEC/EmuELEC #1343: Player 2 don't work in retroarch NES, MD.... Emuelec 4.7

madman80, 2024-03-19:
Controllers are detected by retroarch but player 2 doesn't work. In settings-input seems not to be correctly mapped. 

Steps to reproduce the behavior:
1. Fresh install of Emuelec 4.7 for s905x3.
2. Set game controllers from ES.
3. Add 2 players roms games in NES or MD.
4. Open any of these games (default nestopia core for NES)
5. Play as player 2. Direction or buttons doesn't respond.

Expected behavior
Movement and action buttons of player 2 in the game.

EmuELEC version:
 - Version 4.7

Device:
 - SOC: [e.g. S095x3]
 - Brand: Vontar 
 - Model X3

Additional context
The way to fix is remaping player 2 controller from settings-input (global) and save changes. This bug was not present in Emuelec 4.6.

beta-tester, 2024-03-27:
see here: https://github.com/EmuELEC/EmuELEC/issues/1320

Langerz82, 2024-03-29:
Seems the player2 input config button values need to be set in retroarch.cfg as well to be recognized.
I tested this by using an old config, setting the assignments in RA for player2 and saving.

See:
[image]

Langerz82, 2024-03-29:
I got a test file that fills in the player2 inputs.
 see:
https://github.com/Langerz82/EmuELEC/blob/dev-ra-joy/packages/sx05re/libretro/retroarch/scripts/set_retroarch_joy.sh

madman80, 2024-03-29:
> I got a test file that fills in the player2 inputs. see: https://github.com/Langerz82/EmuELEC/blob/dev-ra-joy/packages/sx05re/libretro/retroarch/scripts/set_retroarch_joy.sh
> 
> save the file to /emuelec/bin, give it chmod +x premissions, then run the script with all controllers on.
> 
> if it corrupts the retroarch.cfg, restore with retroarch.jc.bak.

Can be this script run every time Emuelec boots? Maybe adding it to custom_start.sh? Or you need to run it only once?

Langerz82, 2024-03-30:
ok pretty sure I located the PR thats causing the issue. https://github.com/libretro/RetroArch/commit/e8956418a893e28a953de0a33f152566d04c3f11
rolling back to the previous commit (back about 3 days fixes the issue)
PKG_VERSION="34e28f548aafe13d4c03974bbf0f14af12396dd4"
(in retroarchs package.mk)
I tested it and player2 controls were working on SNES and megadrive cores.

Langerz82, 2024-03-30:
> Can be this script run every time Emuelec boots? Maybe adding it to custom_start.sh? Or you need to run it only once?

@madman80 if you copy emuelecRunEmu.sh to /emuelec/bin then add just the line "set_retroarch_joy.sh" to line:
https://github.com/Langerz82/EmuELEC/blob/6d3e74bf1b35da7782a2d76f9f69241ba977b415/packages/sx05re/emuelec/bin/emuelecRunEmu.sh#L330

It will run every time you run retroarch.

the commands are:
```
cp /usr/bin/emuelecRunEmu.sh /emuelec/bin
nano /emuelec/bin/emuelecRunEmu.sh
```
add "set_retroarch_joy.sh" in one line to line 330.
save close.

This is only a temp solution though, the latest retroarch package bump PR fixes this issue.
If you need to reset the retroarch.cfg, in ES press start go to "emuelec settings >> danger zone" reset the retroarch.cfg, run retroarch, and the retroarch.cfg should restore to the state it was at time of install.

psaezdev, 2024-05-18:
> > > I got a test file that fills in the player2 inputs. see: https://github.com/Langerz82/EmuELEC/blob/dev-ra-joy/packages/sx05re/libretro/retroarch/scripts/set_retroarch_joy.sh
> > > save the file to /emuelec/bin, give it chmod +x premissions, then run the script with all controllers on.
> > > if it corrupts the retroarch.cfg, restore with retroarch.jc.bak.
> > 
> > 
> > Can be this script run every time Emuelec boots? Maybe adding it to custom_start.sh? Or you need to run it only once?
> 
> @madman80 if you copy emuelecRunEmu.sh to /emuelec/bin then add just the line "set_retroarch_joy.sh" to line: https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/bin/emuelecRunEmu.sh#L327
> 
> It will run every time you run retroarch.
> 
> the commands are:
> 
> ```
> cp /usr/bin/emuelecRunEmu.sh /emuelec/bin
> nano /emuelec/bin/emuelecRunEmu.sh
> ```
> 
> add "set_retroarch_joy.sh" in one line to line 327. save close.
> 
> If you need to disable the call, insert a "#" at the very beginning of the edited line.
> 
> This is only a temp solution though, the latest retroarch package bump PR fixes this issue. If you need to reset the retroarch.cfg, in ES press start go to "emuelec settings >> danger zone" reset the retroarch.cfg, run retroarch, and the retroarch.cfg should restore to the state it was at time of install.
> 
> Oh yeah and the set_retroarch_joy I orginally posted had a few bugs, get the latest one here: https://github.com/Langerz82/EmuELEC/blob/6d3e74bf1b35da7782a2d76f9f69241ba977b415/packages/sx05re/libretro/retroarch/scripts/set_retroarch_joy.sh
> 
> edit: updated instructions and emuelecRunEmu.sh link.

Hi, If I update to beta through emuelec update menu in my 4.7 emuelec, will it solve this issue?

Langerz82, 2024-05-18:
@psaezdev The beta channel version should have the fix, yes.

bitnykk, 2024-06-09:
Yay dudes,, idk what happened in 4.7 but that broke some used-to-work emulators as wii/staturn (check yabanshiro topic in which i explain how i had to script some rom copy in root+each folder to fix) ... but also now i realize 4.7 also broke player 2 controls in most games/emulators. That'd make me want to retro move to 4.6 just for the sake of workable state. 4.7 sounds a broken beta now that i moved into it. That's sad as i'm sure you guys did a lot a of towards releasing it. But it broke many working stuff so what do we do with it ?

madman80, 2024-06-09:
> Yay dudes, idk what happened in 4.7 but that broke some used-to-work emulators as wii/staturn (check yabanshiro topic in which i explain how i had to script some rom copy in root+each folder to fix) [...]

The easiest solution is to update to 4.8TEST.

Langerz82, 2024-06-09:
https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024

bitnykk, 2024-06-09:
> https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024
> 
> FYI - We had 4.7 publicly available in the dev branch for over a month or so but not many tested it. I think in future we will probably use the TEST release for a few months to get more testers involved rather than having users to compile it themselves (which not that many do).

Well i could willingly fire a few tests from times to times. But that means that i'll buy a 2nd micro SD card for a crash test dummy install, and another at workable state staying intact.

beta-tester, 2024-06-10:
> https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024
>
> FYI - We had 4.7 publicly available in the dev branch for over a month or so but not many tested it. I think in future we [...]

thank you for the link. i never noticed this repository before.
i have a spare SD card, 4 joysticks and i own some multi player PSX games i can test with.
maybe next weekend i have time to play around.
i there just the image i can flash to a blank SD card
or is there also an upgrade path, where i can upgrade from SD CArd with 4.6 on it to 4.8TEST via internet?

madman80, 2024-06-10:
> > https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024
> > FYI - We had 4.7 publicly available in the dev branch for over a month or so but not many tested it. I think in future we will probably use the TEST release for a few months to get more testers involved rather than having users to compile it themselves (which not that many do).
> 
> thank you for the link. i never noticed that test repository before. i have a spare SD card, 4 joysticks and i own some multi player PSX games i can test with. maybe next weekend i have time to play around. is there just the image available i can flash to a blank SD card or is there also an upgrade path, where i can upgrade from SD CArd with 4.6 on it to 4.8-TEST via internet?
> 
> PS: It's great when a ready-to-use build is available. i'm reluctant to go to the trouble of installing a complete build environment for every project I want to test.
I was able to update from 4.7 via emuelec/emulationstation menu. You just need to activate beta in that menu and click for update. 4.8TEST should be offered to you.

bitnykk, 2024-06-14:
Been buying a second card to keep stable in first and participate beta with second.
From what i see the 4.8 didn't fix the upper issue with player 2 controls.
In some TG16 games i've tested, the p2 loses some keys while its control conf seems exactlyt identical from p1.
Still, at EE level, both controllers have exact same triggers up as shown activated in upper left corner.
In 4.6 that same config used to work fully and properly, so something wrong changed into 4.7 and 4.6.

Langerz82, 2024-07-10:
> Been buying a second card to keep stable in new one & participate beta with older one. From what i see the 4.8 didn't fix the upper issue with player 2 controls. In some TG16 games i've tested, the p2 loses some keys while its control conf seems exactlyt identical from p1. Still, at EE level, both controllers have exact same triggers up as shown activated in upper left corner. In 4.6 that same config used to work fully and properly, so something wrong changed into 4.7 and 4.8.

For 4.8 TEST Try and reset your retroarch.cfg settings in ES Menu Settings. Then re-try.

beta-tester, 2024-07-16:
> thank you for the link. i never noticed that test repository before. i have a spare SD card, 4 joysticks and i own some multi player PSX games i can test with. maybe next weekend i have time to play [...]

i upgraded from EE4.6 to EE4.8 (4.8-TEST-04212024) and two joysticks were working as expected for the following emulators:
- PSX
- SNES
- NES
- 2600
- Arcade (MAME-2003-Plus)

for this emulator i had to reconfigure the inputs:
- AdvanceMAME

Langerz82, 2024-09-01:
@beta-tester AdvanceMame is a different beast as it uses auto gamepad.

Otherwise 4.8TEST and onwards should fix this issue.

closed 2024-09-01 by Langerz82

andyvi23, 2024-11-04:
hi @Langerz82  , I had download the last file https://github.com/Langerz82/EmuELEC/blob/6d3e74bf1b35da7782a2d76f9f69241ba977b415/packages/sx05re/libretro/retroarch/scripts/set_retroarch_joy.sh and downloaded to /emuelec/bin .
And then copy : cp /usr/bin/emuelecRunEmu.sh /emuelec/bin 
And the last I will replace ( remove all lone 327 ) the line 327 with content  :      RUNTHIS='${TBASH} gmloader.sh "${ROMNAME}" --controllers="${CONTROLLERCONFIG}"'     to  content "set_retroarch_joy.sh" 
It show like 
Before:
 "gmloader")
            set_kill_keys "gmloader"
            CONTROLLERCONFIG="${arguments#*--controllers=*}"
            RUNTHIS='${TBASH} gmloader.sh "${ROMNAME}" --controllers="${CONTROLLERCONFIG}"'   <-- this line 327
        ;;
After:
 "gmloader")
            set_kill_keys "gmloader"
            CONTROLLERCONFIG="${arguments#*--controllers=*}"
            set_retroarch_joy.sh      <--- this line 327
        ;;

It right or wrong , Thanks.

Langerz82, 2024-11-04:
set_retroarch_joy.sh only modifies the retroarch joypad configuration. I dont think be used to modify controls for gmloader. I seperate file would have to coded for it, but I don't know what the input config file should look like.