EmuELEC/EmuELEC #1345: EmuElec avec 4 players games

batpoigne, 2024-03-26:
System : BeeLink GT King with 4.7 release

Hello,
For the moment, the only systems for me compatible with 4 players games is Arcade (like "simpsons" for exemple)

I have try to play some 4 players game like bomberman ou micromachines in megadrive, supernintendo ou PS1 but it doesn't works...

Can you tell me if it's possible and what is the way to do that ?

Thanks
Batpoigne

beta-tester, 2024-03-27:
see here: https://github.com/EmuELEC/EmuELEC/issues/1320

Langerz82, 2024-03-30:
Temp fix and more info here:
https://github.com/EmuELEC/EmuELEC/issues/1343

closed 2024-03-30 by Langerz82

batpoigne, 2024-04-02:
Hello,
I have try the option explained in the comment : OK for some systems like PS1 ou megadrive...but for the moment, I don't see any option to do this with supernintendo for exemple...I think it's a problem with emulators perhaps, I have tried some options but no way to configure a 4 players party with theis system.
Any idea ?
THANKS