EmuELEC/EmuELEC #1350: not working : "YabasanshiroSA" for Sega Saturn games in Emuelec 4.7

skiiingray, 2024-04-11:
**Describe the bug**
one error as below when it's run Sega Saturn rom, chd-format by the default emulator "yabasanshiroSA"  of emuelec 4.7
"
File not found:
Cannot initialize ISO-File Virtual Drive
revoke thread
use thread
Cannot initialize Game
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be number, but is null
Aborted
"
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Sega Saturn game collection'
2. Click on 'any game rom of chd format either US or JP version'
3. Leave all in "Auto" in emulator setting by default - stand alone Yabasanshiro / yabasanshiroSA
4. Drop out after 2 flashing of "Loading Game" screen with the error of log message as above 

**Expected behavior**
it's supposed to either load game to play or stay in YabashiroSA menu 

**Screenshots**
[image]

**EmuELEC version:**
 - Version : 4.7

**Device (please complete the following information):**
SOC: S095x4
Brand: X96
Model : Max+ Ultra

**Additional context**
Instead, it's ok to manually opt in "core : libretro_yabasanshiro" to play roms of games.

bitnykk, 2024-05-25:
Got nearly same log except different game and controller, but same yabasansghiro crash.
Used to worked perfectly fine before i updated to recent Emuelec.

skiiingray, 2024-05-26:
nice comment with an idea as well as what i thought but haven't given it a try yet. thanks for share

Langerz82, 2024-06-01:
@bitnyk can you share the sh file you created? I can look at updating our current sh file.

bitnykk, 2024-06-04:
> @bitnyk can you share the sh file you created? I can look at updating our current sh file.

Well, i defo can share it here, just beware it's a very quick/dirty fix i wrote so that i don't have to redo the copying manually everytime a new game is added into YS/saturn roms folder. This imo would NOT replace a better core level fix. Still you got it :
<<
#!/bin/bash
cd /storage/roms/saturn
cp /storage/roms/bios/mpr-17933.bin .
cp /storage/roms/bios/mpr-18811-mx.ic1 .
cp /storage/roms/bios/mpr-19367-mx.ic1 .
cp /storage/roms/bios/saturn_bios.bin .
cp /storage/roms/bios/sega_101.bin .
cp /storage/roms/bios/stvbios.zip .
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/mpr-17933.bin . {} \;
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/mpr-18811-mx.ic1 . {} \;
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/mpr-19367-mx.ic1 . {} \;
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/saturn_bios.bin . {} \;
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/sega_101.bin . {} \;
find . -maxdepth 1 -type d -exec cp /storage/roms/bios/stvbios.zip . {} \;
>>

Regards,
BN

bitnykk, 2024-06-14:
So i've bought a second card to reinstall a stable 4.6 to play from.

Beside this i've updated my old card to 4.8 Beta & i've remove the files my upper script fixed in 4.7.
So i can confirm that presently the same bug (not finding bios files from roms folder) is still up.
It used to work in 4.6 though, so something should have changed in 4.7 and kept in 4.8 breaking it.

skiiingray, 2024-06-15:
noted that and got the same issue in my EE4.8 device

Langerz82, 2024-07-24:
Thanks guys, I have raised the issue with the devs of yabasanshiro and hopefully we can get some kind of fix for the issue.