EmuELEC/EmuELEC #1357: analog devices linux kernel (2019_r2) is not accessible

murinrtc, 2024-04-16:
I have been facing the following issue since 3 months, on cloning the analog devices linux kernel (2019_r2). please provide solution for this.

_git clone -b 2019_R2 https://github.com/analogdevicesinc/linux.git
Cloning into 'linux'...
remote: Enumerating objects: 10911136, done.
remote: Counting objects: 100% (6320/6320), done.
remote: Compressing objects: 100% (2131/2131), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed_

Also, my major problem is in petalinux build.
error message is as follows.

_linux-xlnx-4.19-adi_2019_R2+gitAUTOINC+89a181de04-r0 do_fetch: Fetcher failure for URL: 'git://github.com/analogdevicesinc/linux.git;protocol=https;branch=2019_R2'. Unable to fetch URL from any source._

please provide the solution. i am struck in my project application and in need of help

murinrtc, 2024-04-16:
@shantigilbert can you please suggest any solution.

shantigilbert, 2024-04-16:
First, what does this have to do with EmuELEC? 
second why are you tagin me? I have no idea what that repo is.

closed 2024-04-16 by shantigilbert