EmuELEC/EmuELEC #1362: Dualshock 3 can't be connected by bluetooth correctly, they are not saved

psaezdev, 2024-05-10:
**Describe the bug**
Dualshock 3 can't be connected by bluetooth correctly, they are not saved

**To Reproduce**
Steps to reproduce the behavior:
1. Clean installation
2. Connect DS3 with cable and press the center button to connect with emuelec and remove the wire
3. Maybe it will connect maybe not, but now, take another different DS3 and do it the same. It detects it with the wire as controller 2 but after removing the wire it will not be synced with emulec
4. After trying a lot of times and rebooting a lot of times I achieved syncing 3 DS3 controllers but the fourth controller was impossible to achieve.

**Expected behavior**
all my 4 controlers synced by bluetooth

**EmuELEC version:**
 - Version 4.7

**Device (please complete the following information):**
 - SOC:  S095x3
 - Brand: TUREWELL
 - Model T95 max +

**Additional context**
The controllers work perfectly, and they sync correctly with emuelec 4.6. Well, with the issues that got 4.6 that shutdown the controllers after some minutes etc.. but 4.6 remembered the sync of the controllers on first try.

Langerz82, 2024-05-15:
If you need backward compatibility, first SSH into your device then execute the following commands:
```
cd /emuelec/scripts
wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/v4.6/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```
then re-boot, and it should revert to the old bluetooth connect script.

Langerz82, 2024-07-24:
@shantigilbert 
Do you think that maybe we could make another ES Entry to run the old legacy pairing method too?

psaezdev, 2024-07-24:
> If you need backward compatibility, first SSH into your device then execute the following commands:
> 
> ```
> cd /emuelec/scripts
> wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/v4.6/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth
> chmod +x /emuelec/scripts/emuelec-bluetooth
> ```
> 
> then re-boot, and it should revert to the old bluetooth connect script.
> 
> If you want to revert it to the 4.7 script just rename the emuelec-bluetooth file in scripts with .bak:
> 
> ```
> mv /emuelec/scripts/emuelec-bluetooth /emuelec/scripts/emuelec-bluetooth.bak
> ```

> @shantigilbert Do you think that maybe we could make another ES Entry to be able too run the old legacy pairing method too?

Hi guys, that doesnt solve the issue. I even tryed to update to 4.8 beta and I cant pair the dualshock 3

Any workaround?

Langerz82, 2024-07-24:
@psaezdev 
https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#manual-bluetooth-pairing-notes-by-binhex

psaezdev, 2024-07-25:
> @psaezdev https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#manual-bluetooth-pairing-notes-by-binhex
> 
> @psaezdev I have updated the wiki a bit to provide more info on trying to connect controllers, see: https://github.com/EmuELEC/EmuELEC/wiki/Using-PS3-Gamepads

Tryed and didnt work, also the first link seems to be too much complex. 

I'm forced to go back to 4.6.

Please can this be solved? Even in 4.8 beta is broken. Seems to be a very important bug, as ps3 dualshock is one of the most present bluetooth controllers

worstcase-scenario, 2024-07-25:
It is not a bug as I have 4.8 beta and also a PS3-gamepad and it works...

worstcase-scenario, 2024-07-25:
Check the file gamecontrollerdb.txt in /storage/.config/SDL-GameControllerDB and look for the entry "Sony PLAYSTATION(R)3 Controller" in the section "# EmuELEC extra gamepads" at the beginning of the file. Delete that entry, save the file again and do a system-restart of Emuelec. Then try to reconnect your PS3 controller...

Langerz82, 2024-07-25:
@psaezdev 
I made a little script file which is based off of 4.6's bluetooth connect script.

1. First SSH into your device.
2. Enter the following commands:
```
cd /emuelec/scripts
wget https://gist.githubusercontent.com/Langerz82/9799804c3068a9e361602071c0282f87/raw/95707b0b4e7d1cc43334aee59272d7212168de27/bt_helper.sh
chmod +x bt_helper.sh
./bt_helper.sh
```
3. Turn on controller. See if it discovers controller. If it does let it finish running.
4.  Plug Controller into your device, Press Home button, give it a few seconds, then un-plug.
5. It should then auto connect.

psaezdev, 2024-07-25:
> Check the file gamecontrollerdb.txt in /storage/.config/SDL-GameControllerDB and look for the entry "Sony PLAYSTATION(R)3 Controller" in the section "# EmuELEC extra gamepads" at the beginning of the file. Delete that entry (or just put an # in front of it), then save the file again and do a system-restart of Emuelec. Then try to reconnect your PS3 controller...

> @psaezdev I made a little script file which is based off of 4.6's bluetooth connect script.
> 
> 1. First SSH into your device. ( username: root, password: emuelec).
> 2. Enter the following commands:
> 
> ```
> cd /emuelec/scripts
> wget https://gist.githubusercontent.com/Langerz82/9799804c3068a9e361602071c0282f87/raw/95707b0b4e7d1cc43334aee59272d7212168de27/bt_helper.sh
> chmod +x bt_helper.sh
> ./bt_helper.sh
> ```
> 
> 4. Turn on controller. See if it discovers controller. If it does let it finish running.
> 5. Plug Controller into your device, Press Home button, give it a few seconds, then un-plug.
> 6. It should then auto connect.
> 
> If this script works I'll include it in our latest build.

Hi guys, tryed everything, none of them work. I tested with 4 dual shock 3 controllers, and with various wires. It worked perfectly until Emelec 4.6

Dont know what more can I try to solve this

Langerz82, 2024-07-25:
@psaezdev What did the bt_helper output show?

psaezdev, 2024-07-25:
> @psaezdev What did the bt_helper output show?

Hi, read the last pharagraph of my previous message, I added more info.

Here is the log:

```
Scanning for available devices for 60 seconds, please wait... org.freedesktop. DBus. Error.TimedOut: Failed to activat e service 'org.bluez': timed out (service_start_timeou t=25000ms)

Getting pairable devices, please wait.

org.freedesktop.DBus.Error. NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

org.freedesktop. DBus.Error.TimedOut: Failed to activat e service 'org.bluez': timed out (service_start_timeou t=25000ms)

org.freedesktop.DBus.Error. NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

org.freedesktop.DBus. Error. NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the

network connection was broken.

org.freedesktop. DBus.Error.TimedOut: Failed to activat e service 'org.bluez': timed out (service_start_timeou t=25000ms)
```

Langerz82, 2024-07-25:
@psaezdev 
your bluetooth is not working hence why everything is failing.

Try resetting your bluetooth with:
```
system-ctl restart bluetooth.service
````
then check it's still running with
```
system-ctl status bluetooth.service

Post the output here if it does not restart properly and it might give some clues as to why it fails.

psaezdev, 2024-07-25:
> @psaezdev your bluetooth is not working hence why everything is failing.
>
> Try resetting your bluetooth with:
>
> ```
> system-ctl restart bluetooth.service
> ```
>
> then check it's still running with
>
> ``` [...]

system-ctl command not found

Langerz82, 2024-07-25:
@psaezdev shit my bad, it's "systemctl" not "system-ctl"

psaezdev, 2024-07-25:
> @psaezdev shit my bad, it's "systemctl" not "system-ctl"

Well there is something very strange. As you said bluetooth was disabled, I entered emuelec config menu and bluetooth appeared off. I enabled it and restarted emuelec and I still can't pair the controller.

I did your commands and still can't pair it.  And in the emuelec config menu bluetooth is on now.

This is the log;

```
bluetooth.service Bluetooth service

Loaded: loaded (/usr/lib/systemd/system/bluetooth

Active stiverunning) since Thu 2024-07-25 22:

Main PID: 5631 (bluetoothd)

Status: "Running" Tasks: 1 (limit: 3719)

Memory: 744.0K CGroup:

/system.slice/bluetooth.service 5631 /usr/lib/bluetooth/bluetoothd

Jul 25

22:12:16

EMUELEC bluetoothd[5631]: src/plugin.c

Jul 25

22:12:16

EMUELEC bluetoothd [5631]: src/plugin.c

Jul 25 22:12:16 Jul

EMUELEC bluetoothd[5631]: src/plugin.c EMUELEC bluetoothd[5631]: src/plugin.c

25 22:12:16

EMUELEC bluetoothd[5631]: src/plugin.c

Jul 25 Jul 25

22:12:16

22:12:16

EMUELEC bluetoothd[5631]: profiles/net

EMUELEC bluetoothd [5631]: src/plugin.c

Jul 25 22:12:16

EMUELEC bluetoothd [5631]: Bluetooth ma

Jul 25

22:12:16

EMUELEC bluetoothd [5631]: src/adapter. EMUELEC bluetoothd [5631]: Battery Prov
```

Langerz82, 2024-07-25:
@psaezdev 
That's ok now try running my bt_helper script in SSH.

psaezdev, 2024-07-25:
> @psaezdev That's ok now try running my bt_helper script in SSH. (And post output again).

Now it works with your script! It paires the controller

The normal process doesnt work, but your script works. If i execute it and then try to pair it, it pairs.

Now you know how to solve it? Will it be solved in next emuelec version? Beta or stable?

This is the log, and thank you very much!

```
Scanning for available devices for 60 seconds, please

wait...

Getting pairable devices, please wait...

Icon: input-gaming

Found MAC: E0:AE:5E:5F:41:30

Name: <unknown>

Found controller E0: AE:5E:5F:41:30 Name: <unknown>, tr

usting...

Trusted <unknown>, quick pause, then pairing...

org.bluez.Error.AuthenticationFailed: Authentication F ailed

Icon: input-gaming

Found MAC: E0:AE:5E:5F:41:30

Name: <unknown>

Icon: input-gaming

Found MAC: E0:AE:5E:5F:41:30

Name: <unknown>

Icon:input-gaming

Found MAC: E0:AE:5E:5F:41:30

Name: <unknown>

Icon:input-gaming

Name: <unknown>

Found MAC: E0:AE:5E:5F:41:30
```

Langerz82, 2024-07-25:
@psaezdev 
Good to know. I'll talk to shantigilbert and see if we can include an option for it in the ES menu for future versions. :)

psaezdev, 2024-07-25:
> @psaezdev Good to know. I'll talk to shantigilbert and see if we can include an option for it in the ES menu for future versions. :)

What was the problem?

Langerz82, 2024-07-26:
Problem? Oh I just wanted to make sure the script works ok before making such an update to EmuELEC. You will most likely see this feature to switch pairing methods in 4.8 or 4.9.

See PR:
https://github.com/EmuELEC/emuelec-emulationstation/pull/95

psaezdev, 2024-07-26:
> Problem? Oh I just wanted to make sure the script works ok before making such an update to EmuELEC. You will most likely see this feature to switch pairing methods in 4.8 or 4.9.
> 
> See PR: [EmuELEC/emuelec-emulationstation#95](https://github.com/EmuELEC/emuelec-emulationstation/pull/95)

4.8 if possible please.

Bytheway, I understand that the version that works with ps3 dualshock is old, and has been replaced with a new version that doesnt work with DS3. So users with DS3 will need to activate an option in the emuelec menu to use the old version. Why the modern bluetooth pairing version doesnt work with such a top sale and good controller like DS3? Whouldnt be better to solve the issue and make the new blutetooth pairing version make work with DS3 instead of making necessary to enable a switch on the menu? Is not possible to add DS3 compatibility with the new version? 

Thank you for the explanations. Is just curiosity.

Langerz82, 2024-07-26:
@psaezdev issue is I do not have a DS3 I can test with. If you want to test out the new script and debug pm me in discord when your free and I can go through the steps with you.

Langerz82, 2024-07-31:
Test fix details here:
https://github.com/EmuELEC/EmuELEC/pull/1375

psaezdev, 2024-07-31:
> @psaezdev - I updated the latest script but it needs testing:
> 
> Someone with a Dualshock 3 Controller that is having the same issue try this:
> 
> 1. SSH into your device. (username: root, password: emuelec)
> 2. Copy and paste one line at a time the following commands:
> 
> ```
> cd /emuelec/scripts
> wget https://gist.githubusercontent.com/Langerz82/8558cbfad835d43f3a14719c25dbdeec/raw/ee63c8f730540080c34964ae9becb63dce8794e9/emuelec-bluetooth-agent
> chmod +x /emuelec/scripts/emuelec-bluetooth-agent
> wget https://gist.githubusercontent.com/Langerz82/3033bffa51476862e3cffd62262c0833/raw/6fe160e58527f8c554d27392b1004541dd8319d9/emuelec-bluetooth
> chmod +x /emuelec/scripts/emuelec-bluetooth
> ```
> 
> Now you have the updated scripts, reboot the device and try pairing the devices like you would normally see: https://github.com/EmuELEC/EmuELEC/wiki/Using-PS3-Gamepads

Hello, yes, I tryed always with bluetooth ON since I noticed it was off.

I did your new commands and after doing those commands, if I forgot the DS3 and try to pair it again it doesn't pair anymore. Its broken again with your new code.

I passed also your bt_helper.sh and it doesn't work anymore since I executed your new code quoted here.

This is the output of your script:

```
EMUELEC:/emuelec/scripts # ./bt_helper.sh
Scanning for available devices for 60 seconds, please wait...
Getting pairable devices, please wait...
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:1B:FB:EE:85:A8    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

[... 402 lines cut ...]
Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:1B:FB:EE:85:A8    Name: Sony PLAYSTATION(R)3 Controller
EMUELEC:/emuelec/scripts #
```

Langerz82, 2024-08-01:
Do:
```
rm /emuelec/scripts/emuelec-bluetooth-agent
rm /emuelec/scripts/emuelec-bluetooth
```
reboot and try the bt_helper.sh

psaezdev, 2024-08-01:
> cd /emuelec/scripts
> wget https://gist.githubusercontent.com/Langerz82/9799804c3068a9e361602071c0282f87/raw/95707b0b4e7d1cc43334aee59272d7212168de27/bt_helper.sh
> chmod +x bt_helper.sh
> ./bt_helper.sh

tryed, same problem

can you try with your changes and another controller?

this is very frustrating :(

```
Scanning for available devices for 60 seconds, please wait...
Getting pairable devices, please wait...
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:input-gaming
Found MAC: E0:AE:5E:5F:41:30    Name: Sony PLAYSTATION(R)3 Controller
Icon:input-gaming
Found MAC: 00:06:F5:B6:56:BE    Name: Sony PLAYSTATION(R)3 Controller
```

Langerz82, 2024-08-01:
couple of things to try:
run:
```
ps -a | grep bluetooth
```
check if any emuelec-bluetooth-agent is running.

Also try forgetting the controller bluetooth device in ES Controller settings,
run bt_helper and then when it ran, also connect controller via usb.

worstcase-scenario, 2024-08-01:
Have you tried resetting your PS3-controller? Use a tooth-pick and pit it inside the little hole on the back, press it for some seconds.

psaezdev, 2024-08-01:
> Have you tried resetting your PS3-controller? Use a toothpick and put it inside the little hole on the back, press it for some seconds.

yes, tryed it

> couple of things to try: run:
> 
> ```
> ps -a | grep bluetooth
> ```
> 
> check if any emuelec-bluetooth-agent is running.
> 
> Also try forgetting the controller bluetooth device in ES Controller settings, run bt_helper and then when it ran, also connect controller via usb.

this is what I see if I put that on ssh:

```
EMUELEC:~ # ps -a | grep bluetooth
 3874 root      0:00 /usr/lib/bluetooth/bluetoothd
 4734 root      0:00 grep bluetooth
```

Yes I always do that thing, I always forget the devices if present (usually not now :D) and then I run the helper and connect via USB. I also tryed without running bt_helper.

Have you tryed with your new commit but with other bluetooth controllers? do they sync? 

Previously to your new commit, simply using your bt_helper, was possible for me to sync the DS3, now I'm stuck without the possibility to do it.

Langerz82, 2024-08-01:
emuelec-bluetooth-agent should not be intefering with the process if the file is not present. Can you list the files in /emuelec/scripts?
```
ls /emuelec/scripts
```
can you also post the output of your default emuelec-bluetooth file?
```
cat /usr/bin/emuelec-bluetooth
```

psaezdev, 2024-08-01:
> @psaezdev emuelec-bluetooth-agent should not be intefering with the process if the file is not present. Can you list the files in /emuelec/scripts?
> 
> ```
> ls /emuelec/scripts
> ```
> 
> can you also post the output of your default emuelec-bluetooth file?
> 
> ```
> cat /usr/bin/emuelec-bluetooth
> ```
> 
> Also try running this:
> 
> ```
> systemctl restart bluetooth
> ```

the list:

`EMUELEC:~ # ls /emuelec/scripts
bt_helper.sh               bt_helper.sh.6
bt_helper.sh.1             emuelec-bluetooth-agent.1
bt_helper.sh.2             emuelec-bluetooth.1
bt_helper.sh.3             emuelec-bluetooth.2
bt_helper.sh.4             emuelec-bluetooth.3
bt_helper.sh.5             put_user_scripts_here`

the output:

```
EMUELEC:~ # cat /usr/bin/emuelec-bluetooth
cat: can't open '/usr/bin/emuelec-bluetooth': No such file or directory
```

I also tryed to restart bluetooth with no results

I'm very sorry but I'm going vacations and I need emuelec working, so I'm forced to format my sd and install the 4.6 version again, so can't help anymore :(

Langerz82, 2024-08-01:
@psaezdev Ah no worries thanks for trying and testing. I will make an option in ES to run the old version of connecting, as well as the current one. The custom bluetooth-agent PR seems to have issues so I will leave it out for now. It will probably be include in 4.8 when it is released not sure depends on lead admin, if he wants to include it or not when 4.8 is released.

psaezdev, 2024-08-01:
@Langerz82 thank you, maybe @shantigilbert has a Dualshock 3 controller and can test it. It's one of the most sold controllers on all times, for sure a lot of people will be very happy to be able to use it with emuelec 4.8

Thank you

Langerz82, 2024-08-22:
With the latest dev version PR https://github.com/EmuELEC/EmuELEC/pull/1374 will give the user the option to switch bluetooth pairing to legacy mode in the ES options.

It will probably be included in EmuELEC 4.9, maybe 4.8 depending on what shantigilbert decides to include in the next release.

shantigilbert, 2024-08-23:
> @Langerz82 thank you, maybe @shantigilbert has a Dualshock 3 controller and can test it. It's one of the most sold controllers on all times, for sure a lot of people will be very happy to be able to use it with emuelec 4.8
> 
> Thank you

Sorry I do not have a DS3 controller, and they are a PITA to use on EmuELEC.

Langerz82, 2024-09-01:
Legacy BT Pairing method in latest version, update to 4.9 when it comes out and you will have the option to switch to legacy pairing in ES.

closed 2024-09-01 by Langerz82