EmuELEC/EmuELEC #1369: Emuelec 4.7 on S905y4 - dtb not working

OOKAMIFR, 2024-06-25:
Hello 😊 

First,  thanks for your app update 😊

I have the following device I would like to run with Emuelec 

Speedrun M98 stick 
S905y4 
Mali G31

Ram 4gb
Rom 64GB 

I First tried Corelec with the generic image and the S905y4 4gb dtb and confirm it work perfect 

After this, I prepared Emuelec with the generic Amlogic image and choose s4-s905y4 image

The Transpeed logo appears but it reboot to the logo

Also tested with S905y4 dtb from Corelec and having the same issue 

Hope my explanation is complete and many thanks for your help 🙏 😊

Langerz82, 2024-06-29:
What was the CoreELEC image version? We use V20.3 so if that version does not work on your device, neither will this.

OOKAMIFR, 2024-06-29:
> What was the CoreELEC image version? We use V20.3 so if that version does not work on your device, neither will this.

Hello 😊 
Thanks for your reply 
From official site 
Stable version 21.10
When choosing device - S905y4 generic

SamuEDL, 2024-06-30:
At the moment, we do not support the S905y4 since the kernel we use does not support it. Since it was implemented in amlogic-ne, and for now we continue using coreelec's amlogic-ng kernel.

OOKAMIFR, 2024-07-01:
Ok 😊 
I just check Corelec Github and see S905y4 dtb is present on the 20.3 Nexus

In this case I will test this release and if it work test Emuelec with Corelec dtb 

If not I will simply have to wait for the release 

All my thanks for your help and time 😊

Do I change title with SOLVED word  ?

worstcase-scenario, 2024-07-09:
You can just close your issue..

closed 2024-07-09 by Langerz82