EmuELEC/EmuELEC #1370: L2 and R2 buttons do not work in PSX games on emuelec 4.7

alonsocr23, 2024-07-08:
I have tried L2 and R2 buttons in the PSX emulator in both PCSX Rearmed and Swanstation and neither L2 nor R2 buttons work, I have tried it even with 2 different types of controllers and neither. The only way I have managed to get them to work is by configuring the buttons from the retroarch menu.
All this I have done in Emuelec 4.7 and in a Tvbox HK1box S905x3. 
I hope it helps you.

Kez66, 2024-07-09:
Hi,
I had the same problem as you. I updated to 4.8 and that fixed this button error. On the Tomb Raider for example. 

https://github.com/EmuELEC/EmuELEC/pull/1367

:)

alonsocr23, 2024-07-09:
Hi, I've been looking to see if someone posted that problem and I didn't see it, that's why I opened the topic. 
One question, how can I test version 4.8 (even if it is beta version) to check if they have fixed the bugs of version 4.7? 
Best regards.

Kez66, 2024-07-09:
I posted here just like you in April and reported the same problem as you. As I remember, it was Shantigilbert who shared with me a link to Emuelec 4.8 TEST. Since this update I have no longer had this button problem on PSX games. I still have this version 4.8 TEST in ISO format. But I don't know how to share it.

worstcase-scenario, 2024-07-09:
You can also update to 4.8 test via the menu. Just make sure you choose beta as an option.

Langerz82, 2024-07-10:
Issue fixed in comments.

closed 2024-07-10 by Langerz82