EmuELEC/EmuELEC #1376: Red screen on Dreamcast games, version 4.8 TEST

alonsocr23, 2024-07-30:
Hi, testing the Emuelec version 4.8 TEST I have found that in the game Crazy Taxi both European and USA versions (Dreamcast emulator, flycast) do not work, the red error screen appears.

[image]

I am testing it on HK1 Box S905x3.
I hope you find it helpful, if I detect any more bugs I will let you know. Best regards.

Kelvfimer, 2024-07-30:
Pls can you scroll down the screen to see the error message?

alonsocr23, 2024-07-30:
Tomorrow i will do it😉

Kelvfimer, 2024-07-30:
I'm pretty sure it would fixed as I updated the core yo latest version 2 days ago. I will also test tomorrow.

alonsocr23, 2024-07-31:
Here are the complete screenshots of all red screen errors
[image]
[image]
[image]

skiiingray, 2024-07-31:
looked like it lacked certain bios (dc_boot.bin) to load. you may need to place necessary dc bios to /roms/bios/dc/.

alonsocr23, 2024-07-31:
> looked like it lacked certain bios (dc_boot.bin) to load. you may need to place necessary dc bios to /roms/bios/dc/.
> 
> 

Well, I have already checked that and I did have the BIOS in /roms/bios/dc, and they are the same BIOS that I used in version 4.7 and it does work there, so I don't know what the fault is.

skiiingray, 2024-07-31:
seems to me, it worked well for playable emulation w/o issue given EE 4.8 test build @ 905X4 device.

my log file shown below. FYI.  you may give a shot with the attached bios file. 
[dc_boot.zip](https://github.com/user-attachments/files/16443092/dc_boot.zip)

EmuELEC Run Log

##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.8-TEST-04212024 (Amlogic-ng.aarch64)
PLATFORM: dreamcast
ROM NAME: /storage/roms/dreamcast/Crazy Taxi (USA).chd
BASE ROM NAME: Crazy Taxi (USA).chd
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/dreamcast/Crazy Taxi (USA).chd
2nd Argument: -Pdreamcast
3rd Argument: --core=flycast
4th Argument: --emulator=libretro
Full arguments: /storage/roms/dreamcast/Crazy Taxi (USA).chd -Pdreamcast --core=flycast --emulator=libretro --controllers=-p1index 0 -p1guid 0300ef68100800000100000010010000 -p2index 1 -p2guid 0300ef68100800000100000010010000 -wheel  -autosave 1 -state_slot 0
Run Command is:
retroarch -v -L /tmp/cores/flycast_libretro.so --set-shader /tmp/shaders/crt/crt-potato-BVM.glslp --config /storage/.config/retroarch/retroarch.cfg /storage/roms/dreamcast/Crazy Taxi (USA).chd
Emulator Output is:
[INFO] RetroArch 1.18.0 (Git c0692dbd1b)
[INFO] === Build =======================================
[INFO] CPU Model Name: Amlogic S905X4 rev d

[INFO] Capabilities: ASIMD 
[INFO] Version: 1.18.0
[INFO] Git: c0692dbd1b
[INFO] Built: Apr 21 2024
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/flycast_libretro.so"
[INFO] [Overrides]: Redirecting save file to "/storage/roms/dreamcast/Crazy Taxi (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/savestates/dreamcast/Crazy Taxi (USA).state".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Core-specific remap found at "/storage/.config/retroarch/config/remappings/Flycast/Flycast.rmp".
[INFO] [Environ]: GET_LOG_INTERFACE.
[libretro INFO] shell/libretro/libretro.cpp:321 N[BOOT]: retro_init
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: SET_KEYBOARD_CALLBACK.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[libretro INFO] core/hw/mem/addrspace.cpp:431 N[VMEM]: Info: nvmem is enabled
[libretro INFO] core/hw/mem/addrspace.cpp:463 N[VMEM]: BASE 0x7f829d0000 RAM(16 MB) 0x7f8e9d0000 VRAM64(8 MB) 0x7f869d0000 ARAM(2 MB) 0x7fa29d0000
[INFO] [Content]: Content loading skipped. Implementation will load it on its own.
[libretro INFO] shell/libretro/libretro.cpp:1993 N[BOOT]: retro_load_game: /storage/roms/dreamcast/Crazy Taxi (USA).chd
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/bios".
[INFO] [Environ]: SAVE_DIRECTORY: "/storage/roms/dreamcast/".

alonsocr23, 2024-07-31:
Maybe the problem is only in the s905x3. If you want send me the BIOS files and I'll try with those.

Kelvfimer, 2024-07-31:
@alonsocr23 I tested and it is working on my side. I'm using odroid n2+. Per logs I would say is a BIOS issue.

Kelvfimer, 2024-07-31:
@alonsocr23 tested by a colleague in discourse same board as you and no issues. Maybe you need to reset to default binaries and scripts

Langerz82, 2024-08-01:
```
[INFO] RetroArch 1.18.0 (Git c0692dbd1b)
[INFO] === Build =======================================
[INFO] CPU Model Name: Amlogic S905X4 rev d

[INFO] Capabilities: ASIMD
[INFO] Version: 1.18.0
[INFO] Git: c0692dbd1b
[INFO] Built: Apr 21 2024
```
@Kelvfimer is it the same Retroarch version and build version as you? Can you check if possible. If it's the same then its a user issue not a bug per say.

Kelvfimer, 2024-08-01:
@Langerz82  yes it was tested by AlienatedDiego S905x3 , 4.8 test oficial RA 1.18 .

Besides I had a similar issue when I upgraded Flycast that suddenly the Bios were not ok.  I had to change them. I'm trying to speak to @alonsocr23 via Discord I think he is Alonsocr85 to check the bios files but no answer.

alonsocr23, 2024-08-01:
> @Langerz82  yes it was tested by AlienatedDiego S905x3 , 4.8 test oficial RA 1.18 .
>
>
>
> Besides I had a similar issue when I upgraded Flycast that suddenly the Bios were not ok.  I had to change [...]

@Kelvfimer  I've already written to you on discord, we're in touch there😉

Kelvfimer, 2024-08-01:
It is solved it was an issue on the rom and the Bios were not ok. @alonsocr23 pls close it

closed 2024-08-01 by alonsocr23

stgoddv, 2024-10-13:
> It is solved it was an issue on the rom and the Bios were not ok. @alonsocr23 pls close it

Hi Jelvfimer. Can you please elaborate? What was the issue specifically? How did you solved it? That could help other people..
 
I'm having exactly the same issue with the redscreen, only with dreamcast and the "missing bios" message. I'm using Emuelec 4.7. 

Thanks

worstcase-scenario, 2024-10-13:
You need to try another BIOS file if you get that kind of error.