EmuELEC/EmuELEC #1377: Is it possible to add supermodel on EmuELEC?

25minutes, 2024-07-31:
**Is your feature request related to a problem? Please describe.**
I just found supermodel (Sega Model 3 emulator) has an [ARM friendly branch](https://github.com/DirtBagXon/model3emu-code-sinden). Is it possible to make it work on EmuELEC?

**Describe the solution you'd like**
make supermodel work on EmuELEC

**Describe alternatives you've considered**
<none>

**Additional context**
<none>

Kelvfimer, 2024-07-31:
@25minutes I as able to compile it however it states the binary is x86x64

EMUELEC:/emuelec/bin/bin # ./supermodel
-sh: ./supermodel: cannot execute binary file: Exec format error

EMUELEC:/emuelec/bin/bin # file supermodel
supermodel: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=71ecfdf6f5604faa92633645faa7fc5dc6684aa1, for GNU/Linux 3.2.0, stripped

Maybe @emuelec can take a look into this. I think is an interesting emulator who knows if it's very GPU demanding.

Kelvfimer, 2024-07-31:
Ok I see the issue would be here ARCH = -march=native https://github.com/DirtBagXon/model3emu-code-sinden/blob/main/Makefiles/Rules.inc I'll try setting it ARM

Kelvfimer, 2024-07-31:
as I'm compiling in Intel I do not know how to set to use ARM for my odroidn2+ 

that's the change I did sed -i "s|ARCH = -march=native|ARCH = -march=armv8-a|g" Makefiles/Rules.inc

the error and the switch I can use.

. cc1: error: bad value ('armv8-a') for '-march=' switch
cc1: note: valid arguments to '-march=' switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake tigerlake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 native

25minutes, 2024-08-01:
Thanks for looking into this. 
I'm not a developper and have no coding knowledge. I see the following from the author's note, just in case you missed it. Otherwise I have no idea how to compile it.

> For arm based systems, clone the arm branch:
> git clone --single-branch --branch arm https://github.com/DirtBagXon/model3emu-code-sinden.git

Kelvfimer, 2024-08-01:
We will not be able to add it. It depends on Glu and we don't have support for it.We are sorry :(.

We will take in account in future releases.

 Pls if you can close the issue we appreciate ir.

closed 2024-08-02 by 25minutes