EmuELEC/EmuELEC #1382: Unable to configure 8bitdo M30 controller in Emuelec 4.7

alonsocr23, 2024-08-14:
I have 2 8bitdo M30 controllers and updated to their latest version. The first thing is that it is difficult to link but once I link it when I try to configure the buttons I only set the crosshead up and no longer reacts to the rest or crosshead or buttons. 
When in the Main Menu or on the home screen it scrolls down without touching anything.
I am testing it on a HK1 box S905x3 with Emuelec 4.7.
I have also tested it on the Emuelec 4.8 TEST version and the same thing happens.

Langerz82, 2024-08-18:
You might have to change the input mode. For more Info see:
https://github.com/EmuELEC/EmuELEC/issues/1323

alonsocr23, 2024-08-18:
> You might have to change the input mode. For more Info see:
> 
> https://github.com/EmuELEC/EmuELEC/issues/1323
> 
> 

Well, my 8bitdo M30 model doesn't use the 2.4Ghz receiver. I turn on the controller in D-Input mode (Start + B) and then hold down the Bluetooth button and the error remains the same.

Langerz82, 2024-08-18:
Try this:

> @shantigilbert This is resolved. There is mode change method in M30 2.4G pad.(I couldn't find this in manual or 8bitdo homepage)
> 
> Home+up (5sec) : x-input mode Home+left (5sec) : switch mode Home+down (5sec) : d-input mode
> 
> Select+up (5sec) : d-pad only mode Select+left (5sec) : dpad as left analog Select+right (5sec) : dpad as right analog Select+down (5sec) : swap A/B, X/Y
> 
> 1. change x-input mode
> 2. then M30 is connected with "X-box 360 pad"
> 3. change dpad only mode
> 4. keymap in emulstation
> 5. D-pad is now recognized as "Hat0 up/down/left/right" correctly.
> 
> Thanks.

You might need another controller or keyboard to get to the manual controller mapping part in ES.

closed 2024-08-23 by shantigilbert