EmuELEC/EmuELEC #1389: Retrorach logging is allways on

Kelvfimer, 2024-08-30:
**Describe the bug**
Due to this code snippet https://github.com/EmuELEC/EmuELEC/blob/342c5d3c92baa4f93946415d471060f203057987/packages/sx05re/emuelec/bin/emuelecRunEmu.sh#L133

Everytime we run retroarch it is running with flag -v . There is no way to remove the verbosity logging even setting it off in the menu or retroarch.cfg it is changed to ON

Besides the log level in ES doesn't show NONE logs. 

[image]

**To Reproduce**
Run any game and check with ps, top or htop the commnadline

**Expected behavior**
Be able to select if you want verbose logging or not on retroarch or in ES

**EmuELEC version:**
 Latest commit

**Device (please complete the following information):**
 Any device

Kelvfimer, 2024-08-30:
It is also important the variable LOGEMU if not will be stating logs disabled on emuelec.log

https://github.com/EmuELEC/EmuELEC/blob/342c5d3c92baa4f93946415d471060f203057987/packages/sx05re/emuelec/bin/emuelecRunEmu.sh#L486

Besides I see that the sym link with the retroarch log is not created in the /emuelec/logs

Langerz82, 2024-09-04:
@shantigilbert I added a PR for it. Let me know if there is anything I need to adjust.

https://github.com/EmuELEC/EmuELEC/pull/1402

closed 2024-09-12 by shantigilbert