EmuELEC/EmuELEC #1408: Original Sony Sixaxis PS3 Controller is unable to work with bluetooth

joseacr2000, 2024-09-13:
**Describe the bug:**
Original Sony Sixaxis PS3  controller (not dualshock 3) can't be connected by bluetooth. With USB cable works fine

**To Reproduce:**
Steps to reproduce the behavior:
1: Clean installation
2: Push 4: Unplug USB Cable-> Push PS button in controller during 5 seconds Reset Button.
2: Plug USB cable into Original Sony Sixaxis PS3 Controller.
3: Go to Main Menu-> Controller & Bluetooth Settings -> Pair Bluetooth paths automatically
4: Unplug USB Cable-> Push PS button in controller during 5 seconds

I also tried with step 4 unpluging and pluging USB cable twice (just after pushing PS Button), but I didn't work.

**Expected behavior:**
Work with Original Sony Sixaxis PS3 Controller without cable, i.e. using bluetooth.

**EmuELEC version:**
 - 4.7

**Device:**
 - SOC: S905X4
 - Brand: Vontar
 - Model X4

**Additional context**
I tried using the instructions provided in the following ticket:

https://github.com/EmuELEC/EmuELEC/issues/1362

I mean I tested the bt_helper.sh and I didn't work. I also have tested with the following commands with the same poor result:

```
cd /emuelec/scripts
wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/v4.6/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```

Before opening this ticket I treid for a long while and I have not been able to get it to pair even once.

Langerz82, 2024-09-14:
try this:
```
wget https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth-legacy -O /emuelec/scripts/bt_helper.sh
chmod +x /emuelec/scripts/bt_helper.sh
./bt_helper.sh
```
Let it run, turn on bluetooth device, see if it tries to connect, if it does then plug it in, wait a few secs, press home on device, then disconnect and see if it auto-connects
```

joseacr2000, 2024-09-15:
The script you post in the comment above, is the same that appear in the bug https://github.com/EmuELEC/EmuELEC/issues/1362? I tried a lot of time and didn't work. I had a similar result as @psaezdev but without “Found MAC: lines”.

`Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'

Icon:None
org.freedesktop.DBus.Error.InvalidArgs: No such property 'Icon'`

I am upgrading now to 4.7 because I wanted to play this weekend an with 4.6 was possbile.

joseacr2000, 2024-09-15:
I did a try and you fix worked like a charm:

```
login as: root
root@192.168.0.51's password:
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.7 (Amlogic-ng.aarch64)
                                      Machine model: Amlogic
                                     CoreELEC dt-id: sc2_s905x4_4g_1gbit

EMUELEC:~ # cd /emuelec/scripts
EMUELEC:/emuelec/scripts # wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/dev/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth-legacy
--2024-09-15 14:32:08--  https://raw.githubusercontent.com/EmuELEC/EmuELEC/dev/packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth-legacy
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2004 (2.0K) [text/plain]
Saving to: ‘emuelec-bluetooth-legacy’

emuelec-bluetooth-l 100%[===================>]   1.96K  --.-KB/s    in 0s

2024-09-15 14:32:08 (4.62 MB/s) - ‘emuelec-bluetooth-legacy’ saved [2004/2004]

EMUELEC:/emuelec/scripts # chmod +x /emuelec/scripts/emuelec-bluetooth-legacy
EMUELEC:/emuelec/scripts # emuelec-bluetooth-legacy
Scanning for available devices for 60 seconds, please wait...
Getting pairable devices, please wait...
Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
Found controller 00:1B:FB:A0:30:29 Name: <unknown>, trusting...
Trusted <unknown>, quick pause, then pairing...
org.bluez.Error.AuthenticationCanceled: Authentication Canceled

Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
Icon:input-gaming
Found MAC: 00:1B:FB:A0:30:29    Name: <unknown>
EMUELEC:/emuelec/scripts # reboot
EMUELEC:/emuelec/scripts #

```

Then, I switched on the gamepad without plugin the USB cable and was connected succesfully. The question are:

1, each time I perform a clean installation it will be needed to perform the instructions provided in this comment: https://github.com/EmuELEC/EmuELEC/issues/1408#issuecomment-2351034384
2. Will this fix work with the following action: Main Menu-> Controller & Bluetooth Settings -> Pair Bluetooth paths automatically... i.e I mean this action call the new version of the script that fixes the issue.

Thanks for your time and cooperation.

Langerz82, 2024-09-16:
@joseacr2000 next release it will have a menu option to switch to the script that was used to make it work. Ideally I'd like to fix our current way of doing it, but I dont have a legit DS3 that has battery life. They all died on me cause I didn't charge them for ages.

Langerz82, 2024-09-16:
If you feel like helping debug our new way of doing it. Do the following:
```
cp /usr/bin/batocera/emuelec-bluetooth /emuelec/bin
sed -i 's/DEBUG = False/DEBUG = True/' emuelec-bluetooth
```
Forget the DS3 wireless Controller in EmulationStation. Press START >> Controller & Bluetooth Settings >> Forget a bluetooth Device >> Select your DS3 Controller.
In SSH:
```
emuelec-bluetooth
```
Turn on your DS3 Controller, see if it tries to connect. If it doesn't paste all the output here. If it says connect cable just do so.

If you want to re-pair your DS3 Controller just type in SSH:
```
emuelec-bluetooth-legacy
```
and use the previous method:
(Let it run, turn on bluetooth device, see if it tries to trust/pair (it needs to trust, but it doesn't need to pair successful), if it does then plug it in, wait a few secs, press home on device, then disconnect and see if it auto-connects).

joseacr2000, 2024-09-16:
I am pleased to help you to debug and collaborate to improve the next release. I made a tried using the commands abvoe to test the new way and it worked. This is the output regarding the requested statemets:

```
EMUELEC:~ # cp /usr/bin/batocera/emuelec-bluetooth /emuelec/bin
EMUELEC:~ # sed -i 's/DEBUG = False/DEBUG = True/' /emuelec/bin/emuelec-bluetooth
EMUELEC:~ # emuelec-bluetooth
running scan for 60 seconds.
> bluetoothctl power on
Changing power on succeeded

> bluetoothctl agent on
> bluetoothctl discoverable on
hci0 new_settings: powered connectable ssp br/edr le secure-conn

hci0 new_settings: powered connectable discoverable ssp br/edr le secure-conn

Changing discoverable on succeeded

> bluetoothctl pairable on
hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn

Changing pairable on succeeded

resuming scan
<  SetDiscoveryFilter success
<  hci0 type 7 discovering on
<  Discovery started
<  [CHG] Controller 43:39:00:00:1F:AC Discovering: yes
[]
<  hci0 type 7 discovering off
<  hci0 type 7 discovering on
<  hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
<  [NEW] Device 00:1B:FB:A0:30:29 00-1B-FB-A0-30-29
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
trusting and pairing device , mac: 00:1B:FB:A0:30:29, icon: input-gaming, paired: False
> bluetoothctl trust 00:1B:FB:A0:30:29
[CHG] Device 00:1B:FB:A0:30:29 Trusted: yes

Changing 00:1B:FB:A0:30:29 trust succeeded

> bluetoothctl pair 00:1B:FB:A0:30:29
Attempting to pair with 00:1B:FB:A0:30:29

hci0 type 7 discovering off

hci0 00:1B:FB:A0:30:29 auth failed with status 0x0e (Disconnected)

hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3

Failed to pair: org.bluez.Error.AuthenticationCanceled

connect cable.
<  [CHG] Device 00:1B:FB:A0:30:29 Trusted: yes
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 type 7 discovering off
<  hci0 00:1B:FB:A0:30:29 auth failed with status 0x0e (Disconnected)
<  hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: no
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: yes
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: no
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: yes
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[... 37 lines cut ...]
BluetoothDevice(mac='F4:FE:FB:5A:66:C1', name='[TV] Samsung 7 Series (43)', alias='[TV] Samsung 7 Series (43)', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-83)
<  [NEW] Device 64:4E:D7:F9:8A:7E 64-4E-D7-F9-8A-7E
[BluetoothDevice(mac='64:4E:D7:F9:8A:7E', name='', alias='64-4E-D7-F9-8A-7E', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-89), BluetoothDevice(mac='F4:FE:FB:5A:66:C1', name='[TV] Samsung 7 Series (43)', alias='[TV] Samsung 7 Series (43)', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-83), BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
[... cut ...] lse, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-83), BluetoothDevice(mac='00:1B:FB:A0:30:29', name='Sony PLAYSTATION(R)3 Controller', alias='Sony PLAYSTATION(R)3 Controller', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
EMUELEC:~ #
Broadcast message from root@EMUELEC (Mon 2024-09-16 15:21:32 CEST):

The system will power off now!
```

Langerz82, 2024-09-16:
@joseacr2000 
It seeems to be trusting the device ok, but fails on pair which is to be expected and it says connect cable which is normal.
Then you need to do following:
Plug the controller into the cable, wait a few secs, press home on device, then disconnect and from then onwards it should auto-connect when turning on.

joseacr2000, 2024-09-16:
I have noticed that now it takes a long time to finish the Emulation Station progress bar (it stays in Supervision for quite a while). I did a top and the emulationstation process is at full CPU for about 5 minutes.

Apart from that change, the only thing I did was to add more zxspectrum roms. Before I had 13K and now it's 60K, but with 13K it was instantaneous and with 60K it takes around 5 minutes, which is insufferable. I don't know if now it is because it has the roms inside subdirectories as you can see in the attached image:

[image]

I comment this to you because I don't know if the new way of pairing can influence or it is for the subject that I comment you of the new roms.

Langerz82, 2024-09-16:
@joseacr2000 
Yeah you probably don't want to nest roms in too many directories, as each folder the gamelist parser probably has to scan its contents. Also this is unrelated to the bluetooth issue, so please open up a new issue for it. But most likely others will say the same thing.

joseacr2000, 2024-09-16:
Yeah you probably don't want to nest roms in too many directories, as each folder the gamelist parser probably has to scan its contents. Also this is unrelated to the bluetooth issue, so please open up a new issue for it. But most likely others will say the same thing.

joseacr2000, 2024-09-24:
I am reopening this ticket because last weekend, I connected my PS3 controller to my PS3 console, and then when I used it again in EmuElec it did not work.

I didn't think too much of it, deleted the controller from the list of linked devices, and tried again to pair it following the steps that made it work the previous time. However, both with the old method based on the lagacy script, and the last one you suggested to try has worked, when the previous time both worked.

I even gave the factory reset to clear the settings (I didn't do a clean install, but cleared all the settings from the EmuleStation graphical environment option) and it didn't work either.

As I was already desperate, I was planning to format everything and reinstall and then put the ROMs unit, but I thought it would be convenient to publish this post to try to solve the problem, since to my understanding, it should work even if you reassign the controller to another device, and then reassign it back to where it was.

joseacr2000, 2024-09-24:
I don't know if we coult take advantage of this ticket instead opening a new one refering it.

worstcase-scenario, 2024-09-24:
I think that you need to reset the controller after you have paired it with the PS3.
Look for a small hole on the back on the controller near the L2 shoulder button. Use an unfolded paper clip to push down the tiny reset button inside the hole.

Then you should be able to pair it again with Emuelec.

joseacr2000, 2024-09-25:
Sorry, I forgot to comment when reopen this bug that I clicked the reset button several times with either short and 15 seconds long press without the desired results.

Langerz82, 2024-09-25:
@joseacr2000 
Reset the controller, In ES select forget bluetooth controller and select the Dual Shock 3 controller record if it's there, then try to re-pair with the script, remember you also need to plug in the cable into the device. If that doesn't work try this:
https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#manual-bluetooth-pairing-notes-by-binhex

joseacr2000, 2024-09-25:
I tried this and it didin't work:

_Reset the controller, In ES select forget bluetooth controller and select the Dual Shock 3 controller record if it's there, then try to re-pair with the script, remember you also need to plug in the cable into the device_

joseacr2000, 2024-09-25:
Using the first method of this post https://github.com/EmuELEC/EmuELEC/issues/1408#issuecomment-2351948808

```
cp /usr/bin/batocera/emuelec-bluetooth /emuelec/bin
sed -i 's/DEBUG = False/DEBUG = True/' /emuelec/bin/emuelec-bluetooth
emuelec-bluetooth
```

It didnt work as you canse below

```
EMUELEC:~ #
login as: root
root@192.168.0.27's password:
Access denied
root@192.168.0.27's password:
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.7 (Amlogic-ng.aarch64)
                                      Machine model: Amlogic
                                     CoreELEC dt-id: sc2_s905x4_4g_1gbit

EMUELEC:~ # cp /usr/bin/batocera/emuelec-bluetooth /emuelec/bin
EMUELEC:~ # sed -i 's/DEBUG = False/DEBUG = True/' /emuelec/bin/emuelec-bluetooth
EMUELEC:~ # emuelec-bluetooth
running scan for 60 seconds.
> bluetoothctl power on
Changing power on succeeded

> bluetoothctl agent on
> bluetoothctl discoverable on
hci0 new_settings: powered connectable ssp br/edr le secure-conn

hci0 new_settings: powered connectable discoverable ssp br/edr le secure-conn

Changing discoverable on succeeded

> bluetoothctl pairable on
hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn

Changing pairable on succeeded

resuming scan
<  SetDiscoveryFilter success
<  hci0 type 7 discovering on
<  Discovery started
<  [CHG] Controller 43:39:00:00:1F:AC Discovering: yes
[]
<  hci0 type 7 discovering off
<  hci0 type 7 discovering on
<  [NEW] Device 54:DF:1B:E0:C6:4E Android TV
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)
<  [NEW] Device 00:1B:FB:A0:30:29 Sony PLAYSTATION(R)3 Controller
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [CHG] Device 54:DF:1B:E0:C6:4E UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [CHG] Device 54:DF:1B:E0:C6:4E UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [CHG] Device 54:DF:1B:E0:C6:4E UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [CHG] Device 54:DF:1B:E0:C6:4E UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [CHG] Device 54:DF:1B:E0:C6:4E UUIDs: 00000000-0000-0000-0000-000000000000
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  [DEL] Device 00:1B:FB:A0:30:29 Sony PLAYSTATION(R)3 Controller
[BluetoothDevice(mac='54:DF:1B:E0:C6:4E', name='Android TV', alias='Android TV', clas='0x0028043c (2622524)', icon='audio-card', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)]
<  hci0 type 7 discovering off
<  hci0 type 7 discovering on
<  hci0 type 7 discovering off
<  hci0 type 7 discovering on
<  [DEL] Device 54:DF:1B:E0:C6:4E Android TV
[]
<  hci0 type 7 discovering off
EMUELEC:~ #
```

joseacr2000, 2024-09-25:
Bluetooth legacy script run into the same lack, i.e. it didin't work:

```
EMUELEC:/emuelec/scripts # emuelec-bluetooth-legacy
Scanning for available devices for 60 seconds, please wait...
Getting pairable devices, please wait...
```
However, following the instructions provided by binhex in the post https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#manual-bluetooth-pairing-notes-by-binhex it was possible to work again with my PS3 bluetooth in EmuElec 4.7

```
EMUELEC:/emuelec/scripts # bluetoothctl
Waiting to connect to bluetoothd...[bluetooth]# Agent registered
default-agentdefault-agent
[bluetooth]# Default agent request successful
power onth]# power on
[bluetooth]# Changing power on succeeded
discoverable onscoverable on
[bluetooth]# hci0 new_settings: powered connectable bondable ssp br/edr le secure-conn
[bluetooth]# hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn
[bluetooth]# Changing discoverable on succeeded
[bluetooth]# [CHG] Controller 43:39:00:00:1F:AC Discoverable: yes
[bluetooth]# [NEW] Device 00:1B:FB:A0:30:29 Sony PLAYSTATION(R)3 Controller
[bluetooth]# Authorize service
[bluetooth]#   1;39m[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): yes
[bluetooth]# [CHG] Device 00:1B:FB:A0:30:29 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
[bluetooth]# [CHG] Device 00:1B:FB:A0:30:29 Class: 0x00000508 (1288)
[bluetooth]# [CHG] Device 00:1B:FB:A0:30:29 Icon: input-gaming
[CHG] Device 00:1B:FB:A0:30:29 Connected: yestroller
[Sony PLAYSTATION(R)3 Controller]# Authorize service
[Sony PLAYSTATION(R)3 Controller]#   1;39m[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): y
[Sony PLAYSTATION(R)3 Controller]# hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 2
[Sony PLAYSTATION(R)3 Controller]# [CHG] Device 00:1B:FB:A0:30:29 ServicesResolved: no
[CHG] Device 00:1B:FB:A0:30:29 Connected: noluetooth
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
[CHG] Device 00:1B:FB:A0:30:29 Connected: yestroller
[Sony PLAYSTATION(R)3 Controller]# Authorize service
[Sony PLAYSTATION(R)3 Controller]#   1;39m[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): y
[Sony PLAYSTATION(R)3 Controller]# hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 2
[CHG] Device 00:1B:FB:A0:30:29 Connected: noluetooth
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
[CHG] Device 00:1B:FB:A0:30:29 Connected: yestroller
[Sony PLAYSTATION(R)3 Controller]# Authorize service
[Sony PLAYSTATION(R)3 Controller]#   1;39m[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): y
[Sony PLAYSTATION(R)3 Controller]# hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 2
[CHG] Device 00:1B:FB:A0:30:29 Connected: noluetooth
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
[CHG] Device 00:1B:FB:A0:30:29 Connected: yestroller
[Sony PLAYSTATION(R)3 Controller]# Authorize service
[Sony PLAYSTATION(R)3 Controller]#   1;39m[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): y
hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 2
[CHG] Device 00:1B:FB:A0:30:29 Connected: no
trust 00:1B:FB:A0:30:29
[bluetooth]# [CHG] Device 00:1B:FB:A0:30:29 Trusted: yes
[bluetooth]# Changing 00:1B:FB:A0:30:29 trust succeeded
[bluetoo00:1B:FB:A0:30:29B:FB:A0:30:29
Attempting to connect to 00:1B:FB:A0:30:29
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connect failed (status 0x04, Connect Failed)
[bluetooth]# Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[bluetooth]# hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
[CHG] Device 00:1B:FB:A0:30:29 Connected: yestroller
[Sony PLAYSTATION(R)3 Controller]# hci0 new_settings: powered connectable bondable ssp br/edr le secure-conn
[Sony PLAYSTATION(R)3 Controller]# [CHG] Controller 43:39:00:00:1F:AC Discoverable: no
[Sony PLAYSTATION(R)3 Controller]#

```
It's quite tricky all this. The first time worked both with both solutions you proposed me. Then, I connect the gamepad to PS3, make a reset and delete the gamepad in ES from the pairid list, and such solutions didn't work at all. However, performing the commands manually, it worked. I don't understand what is happening, but it seems that anywhere is kept some traces of the first connection, and for any reason, scripts are not able to handle this and no longer work.

joseacr2000, 2024-09-25:
Note that the procedure that made it work this time, the one in the previous post is extremely unstable and causes the controller to disconnect every so often, exactly the same behavior as in version 4.6.

As it was very uncomfortable, I deleted the remote from the paired lists, and went back to using the previous method, the one based on the batocera script, i.e. /usr/bin/batocera/emuelec-bluetoot, and this time it worked again (without needing to reset the remote even), having again an exceptional stability.

I pass the output of the script this time to see if it can be fixed for further versions:

```
EMUELEC:~ # cd /emuelec/bin/
EMUELEC:/emuelec/bin #  cp /usr/bin/batocera/emuelec-bluetooth /emuelec/bin
EMUELEC:/emuelec/bin #  sed -i 's/DEBUG = False/DEBUG = True/' /emuelec/bin/emuelec-bluetooth
EMUELEC:/emuelec/bin # ./emuelec-bluetooth
running scan for 60 seconds.
> bluetoothctl power on
Changing power on succeeded

> bluetoothctl agent on
> bluetoothctl discoverable on
hci0 new_settings: powered connectable ssp br/edr le secure-conn

hci0 new_settings: powered connectable discoverable ssp br/edr le secure-conn

Changing discoverable on succeeded

> bluetoothctl pairable on
hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn

Changing pairable on succeeded

resuming scan
<  SetDiscoveryFilter success
<  hci0 type 7 discovering on
<  Discovery started
<  [CHG] Controller 43:39:00:00:1F:AC Discovering: yes
[]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
<  [NEW] Device 00:1B:FB:A0:30:29 00-1B-FB-A0-30-29
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
trusting and pairing device , mac: 00:1B:FB:A0:30:29, icon: input-gaming, paired: False
> bluetoothctl trust 00:1B:FB:A0:30:29
[CHG] Device 00:1B:FB:A0:30:29 Trusted: yes

Changing 00:1B:FB:A0:30:29 trust succeeded

> bluetoothctl pair 00:1B:FB:A0:30:29
Attempting to pair with 00:1B:FB:A0:30:29

hci0 type 7 discovering off

hci0 00:1B:FB:A0:30:29 auth failed with status 0x0e (Disconnected)

hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3

Failed to pair: org.bluez.Error.AuthenticationCanceled

connect cable.
<  [CHG] Device 00:1B:FB:A0:30:29 Trusted: yes
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 type 7 discovering off
<  hci0 00:1B:FB:A0:30:29 auth failed with status 0x0e (Disconnected)
<  hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: no
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR connected eir_len 5
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: yes
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  hci0 00:1B:FB:A0:30:29 type BR/EDR disconnected with reason 3
<  [CHG] Device 00:1B:FB:A0:30:29 Connected: no
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='', alias='00-1B-FB-A0-30-29', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)]
[... 58 lines cut ...]
<  [CHG] Device 00:1B:FB:A0:30:29 Name: Sony PLAYSTATION(R)3 Controller
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='Sony PLAYSTATION(R)3 Controller', alias='Sony PLAYSTATION(R)3 Controller', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 00:1B:FB:A0:30:29 Alias: Sony PLAYSTATION(R)3 Controller
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='Sony PLAYSTATION(R)3 Controller', alias='Sony PLAYSTATION(R)3 Controller', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
<  [CHG] Device 00:1B:FB:A0:30:29 Modalias: usb:v054Cp0268d0000
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='Sony PLAYSTATION(R)3 Controller', alias='Sony PLAYSTATION(R)3 Controller', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
[... cut ...] se, legacy_pairing=False, rssi=0)]
<  hci0 type 7 discovering on
<  hci0 type 7 discovering off
<  No agent is registered
<  [DEL] Controller 43:39:00:00:1F:AC EMUELEC [default]
[BluetoothDevice(mac='00:1B:FB:A0:30:29', name='Sony PLAYSTATION(R)3 Controller', alias='Sony PLAYSTATION(R)3 Controller', clas='0x00000508 (1288)', icon='input-gaming', paired=False, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)]
EMUELEC:/emuelec/bin #

```

Langerz82, 2024-10-01:
I have code for bluetooth agent which can override the security request but the issue is, its skips trusting and pairing so you need to run the pairing script every time you turn on the DS3 controllers.
Anyway, glad to see the default method ended up working for you. In EE 4.8 you can switch legacy and default pairing in ES, without manually fiddling around with files etc.

Langerz82, 2024-10-12:
If anyone still has issues with pairing make sure you reset the controller by putting a pin the hole at the top of the controller and then follow these steps:
https://github.com/EmuELEC/EmuELEC/wiki/Using-PS3-Gamepads

If you still cant get it to pair you may need to manually trust your controller and then try the above.
https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth

I'm closing this issue, if anyone still has problems please open a new issue, as this is quite cluttered.

closed 2024-10-12 by Langerz82