EmuELEC/EmuELEC #1410: L1, L2, R1 and R2 buttons do not under Commdores Amiga PUAE Emulator

joseacr2000, 2024-09-20:
**Describe the error:**
The retropadgame L2 and R2 buttons do not work when using any commodore ammiga rom in the PUAE emulator. By default L2 and R2 are the left mouse button and the right mouse button, but they do not work when you click on it, even though you have clicked before on the key to switch between joystick and mouse (J / M).

**Steps to reproduce the behavior:**
1: Clean the installation.
2. Open any friendly room with PUAE
3: Go to retroach in-game settings and go to Quick Menu->Core Options -> RetroPad Mapping

Then look for the L2 and R2 buttons. By default they are mapped to the left and right mouse buttons respectively:

https://docs.libretro.com/library/puae/

However, for ease of testing, reassign both L2 and R2, for example, to “Toggle virtual keyboard”.

4. Go back and resume to go to the game.
5. Press R2 or L2 during the game and you will see that nothing happens. If it worked as expected, the virtual keyboard should have been displayed on the screen.

**Expected behavior:**
That the action associated with each L2 and R2 key is carried out. For example, with the example described above you would have to show and hide the virtual keyboard on the screen.

**EmuELEC version

4.7
Device:

SOC: S905X4
Brand: Vontar
Model X4

I don't know if it is something related to Emuelec, or it could be a retroach problem, or even the PUAE engine itself. I open it here to see if we can find a solution (L1 and R1 don't work either but all the other buttons and controls work)

worstcase-scenario, 2024-09-20:
I just tested it on 4.8.- test version and there the two buttons are working. Perhaps it's an issue with the trust version... Which version are you on?

joseacr2000, 2024-09-20:
I am working with Emuelec 4.7 and PUAE 5.0.0 and Retroach 1.17. I tried to add the following entries into the EMUELEC:~ # vi /storage/.config/retroarch/config/PUAE/PUAE.opt file and did't work either:

input_player1_mouse_left = "L2"
input_player1_mouse_right = "R2"

The isssue is only related to L2 and R2 buttons (L1 and R1 works fine. I just edited the title)

worstcase-scenario, 2024-09-21:
In my PUAE.opt file there are the following entries, perhaps try adding those:

puae_mapper_l2 = "MOUSE_LEFT_BUTTON"
puae_mapper_r2 = "MOUSE_RIGHT_BUTTON"

Langerz82, 2024-09-21:
4.7 Has a known issue with some controls. You need to update to 4.8TEST and re-test the issue.

joseacr2000, 2024-09-24:
I will make a try in a couple of days with the 4.8TEST. Which revision of the TEST version should I install exactly?

worstcase-scenario, 2024-09-24:
You can get the 4.8.-test-release here: https://github.com/EmuELEC/EmuELEC-tests/releases/tag/v4.8-TEST-04212024

Langerz82, 2024-09-30:
closing issue if the 4.8TEST doesnt work let me know and I will re-open.

closed 2024-09-30 by Langerz82