EmuELEC/EmuELEC #1411: Can´t launch any n64 games

ericitaquera, 2024-09-23:
**Describe the bug**
Hello.

While launching any n64 game from emulation station system lands RetroArch overlay.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch any n64 game from emulationstation

**Expected behavior**
I expected the system was able to run the game.

**EmuELEC version:**
 - Version v4.8-TEST-04212024 (I firstly saw the error on 4.7 but tried the latest TEST)

**Device (please complete the following information):**
 - SOC: Amlogic S905W rev d
 - Brand: 
 - Model: tx9
 - Version: 

**Additional context**

I was able to log in via ssh and get emuelec.log. From it, I got retroarch command line and ran it myself.

Logs available at https://paste.coreelec.org/MansonBlocking

Retroarch commandline returns error:

[ERROR] Failed to open libretro core: "/usr/bin/retroarch"
[ERROR] Error(s): /usr/bin/retroarch: cannot dynamically load executable

Please let me know if I can provide any further useful information.

Langerz82, 2024-09-26:
@ericitaquera How many GB RAM do you have? If it's 1-2GB It might be due to memory constraints. Try running the 32-bit versions in ES, (SELECT >> Advanced Game Options >> Emulator/Core).

Langerz82, 2024-09-28:
Also try removing any Core Options files for Mupen64Plus-Next in SSH do:
```
rm ~/.config/retroarch/config/Mupen64Plus-Next*/*.opt
```
Then re-try and see if it makes any difference.

ericitaquera, 2024-09-28:
It looks like my cpu is no longer supported since 4.3.

Sorry for issuing this issue (sic).

Regards

closed 2024-09-28 by ericitaquera