EmuELEC/EmuELEC #1414: [EE][Input] GameSir T3 Controller incorrect joypad file being used by Retroarch.

Langerz82, 2024-09-30:
**Describe the bug**
Input Device: GameSir T3 Controller.

EmulationStation maps my controller and saves the config to:
/tmp/joypads/Xbox 360 Controller.cfg

However the issue is when I load up Retroarch it loads a different joypad input file:
/tmp/joypads/Microsoft X-Box 360 pad.cfg

What saves when re-mapping:
"Xbox 360 Controller.cfg" file:
```
input_device = "Xbox 360 Controller"
input_driver = "udev"
input_vendor_id = "1118"
input_product_id = "654"
input_r_y_plus_axis = "+4"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_minus_axis = "-3"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_r_btn = "5"
input_save_state_btn = "5"
input_down_btn = "h0down"
input_volume_down_btn = "h0down"
input_r_y_minus_axis = "-4"
input_l_btn = "4"
input_load_state_btn = "4"
input_r_x_plus_axis = "+3"
input_y_btn = "2"
input_x_btn = "3"
input_menu_toggle_btn = "3"
input_b_btn = "0"
input_reset_btn = "0"
input_a_btn = "1"
input_up_btn = "h0up"
input_volume_up_btn = "h0up"
input_select_btn = "6"
input_l3_btn = "9"
input_start_btn = "7"
input_exit_emulator_btn = "7"
input_l_x_plus_axis = "+0"
input_l_y_minus_axis = "-1"
input_enable_hotkey_btn = "6"
input_enable_hotkey = "6"
input_l2_axis = "+2"
input_rewind_axis = "+2"
input_r2_axis = "+5"
input_toggle_fast_forward_axis = "+5"
input_l_y_plus_axis = "+1"
input_r3_btn = "10"
input_fps_toggle_btn = "10"
input_l_x_minus_axis = "-0"

```

What loads into retroarch upon launch:
"Microsoft X-Box 360 pad.cfg" file:
```
input_device = "Microsoft X-Box 360 pad"
input_driver = "udev"
input_vendor_id = "1118"
input_product_id = "654"
input_r_y_plus_axis = "+4"
input_l3_btn = "9"
input_r_x_minus_axis = "-3"
input_l_btn = "4"
input_load_state_btn = "4"
input_start_btn = "7"
input_exit_emulator_btn = "7"
input_l_y_minus_axis = "-1"
input_up_btn = "h0up"
input_volume_up_btn = "h0up"
input_r_y_minus_axis = "-4"
input_a_btn = "1"
input_b_btn = "0"
input_reset_btn = "0"
input_enable_hotkey_btn = "6"
input_down_btn = "h0down"
input_volume_down_btn = "h0down"
input_l_x_plus_axis = "+0"
input_l_y_plus_axis = "+1"
input_r_btn = "5"
input_save_state_btn = "5"
input_r2_axis = "+5"
input_toggle_fast_forward_axis = "+5"
input_r3_btn = "10"
input_fps_toggle_btn = "10"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_x_btn = "3"
input_menu_toggle_btn = "3"
input_select_btn = "6"
input_l_x_minus_axis = "-0"
input_y_btn = "2"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_plus_axis = "+3"
input_l2_axis = "+2"
input_rewind_axis = "+2"
```

cat /proc/bus/input/devices
```
I: Bus=0003 Vendor=045e Product=028e Version=0201
N: Name="Microsoft X-Box 360 pad"
P: Phys=usb-xhci-hcd.0.auto-2.2/input0
S: Sysfs=/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-2/1-2.2/1-2.2:1.0/input/input5
U: Uniq=
H: Handlers=event4 js0 
B: PROP=0
B: EV=20000b
B: KEY=7cdb000000000000 0 0 0 0
B: ABS=3003f
B: FF=107030000 0
```

sdljoytest of input device:
```
EMUELEC:/tmp/joypads # sdljoytest
Sys_InitInput: Compiled with SDL version 2.28.5
Sys_InitInput: Linked   with SDL version 2.28.5
Sys_InitInput: SDL2 joystick subsystem init
Sys_InitInput: Loading /storage/.config/SDL-GameControllerDB/gamecontrollerdb.txt
Sys_InitInput: SDL_GameControllerAddMappingsFromFile() added 2 controller maps
Sys_InitInput: Joystick subsytem - Found 1 joysticks at startup
Joystick 0 name 'Xbox 360 Controller'
Joystick 0 is a game controller
Joystick 0 Axes 06 / Buttons 11 / Hats 01 / Balls 00
Joystick 0 Instance id 0
Joystick 0 Guid 030003f05e0400008e02000001020000
Opened gamepad device index 0 (Xbox 360 Controller)
        axes: 6
     buttons: 11
 instance id: 0
        guid: 030003f05e0400008e02000001020000
     mapping: 030003f05e0400008e02000001020000,Xbox 360 Controller,dpleft:h0.8,rightx:a3,dpright:h0.2,rightshoulder:b5,dpdown:h0.4,righty:a4,leftshoulder:b4,y:b2,x:b3,b:b0,a:b1,dpup:h0.1,back:b6,leftstick:b9,start:b7,lefty:a1,guide:b6,lefttrigger:a2,righttrigger:a5,rightstick:b10,leftx:a0,platform:Linux,crc:f003,
Sys_InitInput: 1 haptic devices detected.
Sys_InitInput: Rumble initialization OK
SDL2: Joytick hotplug supported
Waiting for joystick events. Press CTRL+C to exit.
SDL_JOYDEVICEADDED jdevice.which 00 (Xbox 360 Controller) [DEVICE INDEX]
 Gamepad 00 (Xbox 360 Controller) in use
 Ignoring plugged-in joystick device number 00 (Xbox 360 Controller)
Sys_InitInput: 1 haptic devices detected.
Sys_InitInput: Rumble initialization OK
SDL_CONTROLLERDEVICEADDED cdevice.which 00 (Xbox 360 Controller) [DEVICE INDEX]
 Gamepad 00 (Xbox 360 Controller) already in use
 Ignoring plugged-in gamepad 00 (Xbox 360 Controller)
Sys_InitInput: 1 haptic devices detected.
Sys_InitInput: Rumble initialization OK
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use a GameSir T3 Controller
2. Remap the input in ES.
3. Launch Retroarch
A different joypad profile loads from what is recognised in ES.

**Expected behavior**
Retroarch should be loading the input configuration for the same one ES remaps too.

**EmuELEC version:**
Latest EmuELEC 4.8 TEST

**Device (please complete the following information):**
 - SOC: S922X 4GB RAM
 - Brand: Geelink
 - Model GT King Pro

**Additional context**
ES and sdljoytest recognises it as "Xbox 360 Controller", but  cat /proc/bus/input/devices recognises it as a different name, so we need to figure out why retroarch is using a different joypad input file.

shantigilbert, 2024-09-30:
Retroarch is not using SDL to configure the gamepad, hence why it is different. 
Remember we create the .cfg file with a script that is not related to Retroarch. 

To try and see why it is different 
1.- Try removing all the .cfg files, run Retroarch and configure the controller manually.
2.- Check the resulting .cfg file.

You will end up with a different .cfg that the one we use with the script.

Why this happens? 

Not 100% sure, but if the gamepad uses the same UUID as the original Xbox controller but has a different name then it could mean it is confusing the system.

> I: Bus=0003 Vendor=045e Product=028e Version=0201
> N: Name="Microsoft X-Box 360 pad"

> Joystick 0 name 'Xbox 360 Controller'
> guid: 030003f05e0400008e02000001020000

We use UUID to recognize the gamepad, and Retroarch I believe uses the name (but I am not 100% sure of this). so, when Retroarch loads it uses the name as the Joystick is sending it (N: Name="Microsoft X-Box 360 pad"), but our SDL script is using the other name (Joystick 0 name 'Xbox 360 Controller') to create the config file, so you end up with Retroarch using the "Microsoft X-Box 360 pad.cfg" instead which is for the original controller.

Langerz82, 2024-09-30:
@shantigilbert 

When I removed the "Microsoft X-Box 360 pad" cfg, retroarch loaded my other config fine. So I'm thinking what if we save the guid in the cfg files with a # at the start to comment it out as an actual value (so it's ignored by retroarch), And when we create new cfg mappings we iterate through the joypads and search for the GUID and rename it as a bak file if there is a clash.

This is the code I could change:
https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec-emulationstation/config/scripts/configscripts/retroarch.sh#L380

Langerz82, 2024-10-11:
This will be fixed when the PR is merged so closing.

closed 2024-10-11 by Langerz82