EmuELEC/EmuELEC #1421: emuelec 4.7 on s905x3 arcade/mame issue

Certifr1ed, 2024-10-07:
**Describe the bug**
A clear and concise description of what the bug is.

Most arcade zips run perfectly in arcade folder and i am able to access retroarch with schanwan bluetooth controller. however some games would not open and i got a red error screen, so with these roms i moved to mame folder and they opened and ran perfectly however the bt controller does not work and i can not access retroarch menu or do anything when these mame 2003 roms are in mame folder? 

**To Reproduce**
Steps to reproduce the behavior:
open arcade rom placed in mame folder, runs perfectly but controller does not work.

controller works with arcade roms placed in arcade folder.

**Expected behavior**
i expect the arcade zip/roms to run and have controller support when placed in mame folder

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - 4.7 on s905x3

**Device (please complete the following information):**
 - SOC: s905x3 - Brand: a95x f3 air
 - Model a95x f3
 - Version latest

**Additional context**
please check why arcade roms/zips placed in mame folder run perfectly but controller does nothing and i have to unplug device to fix.

Langerz82, 2024-10-07:
The mame emulator by default is AdvancedMame not libretro so it sound like it's not configuring the input properly.

You will have to SSH in and copy/paste the following output in https://gist.github.com and provide the links on this thread.

try:
```
cat /proc/bus/input/devices
sdljoytest -skip_loop
cat /emuelec/configs/advmame/advmame.rc
````
copy/paste all output in gist, and provide link.

Certifr1ed, 2024-10-07:
Can you clarify what to do once I ssh into my emuelec device using filezilla.
I'm supposed to copy and paste what to  where?

worstcase-scenario, 2024-10-07:
You can't use filezilla for ssh... Use MobaXterm, for example...

Langerz82, 2024-10-07:
Just wondering did you remap the controller buttons manually in EmulationStation? (START >> Bluetooth and Controller Settings >> Remap Controller). AdvanceMame auto joypad scripts require a valid gamecontrollerdb record to work properly.

Certifr1ed, 2024-10-07:
> Just wondering did you remap the controller buttons manually in EmulationStation? (START >> Bluetooth and Controller Settings >> Remap Controller). AdvanceMame auto joypad scripts require a valid gamecontrollerdb record to work properly.
> 

Yes I mapped my controller utilizing the built in method you listed.
To clarify every controls works perfectly in all other cores and emulators
It's only the mame folder placed games where I can not control anything.
Any way to manually edit controller settings in mame emulator?

Langerz82, 2024-10-07:
Without those log information I cannot help you and know exactly what the issue or what to fix.

Certifr1ed, 2024-10-07:
> Without those log information I cannot help you and know exactly what the issue or what to fix.

I will get that info when I get home from work a bit later. Thank you

Certifr1ed, 2024-10-08:
> The mame emulator by default is AdvancedMame not libretro so it sounds like it's not configuring the input properly.
> 
> You will have to SSH in and copy/paste the following output in https://gist.github.com and provide the links on this thread.
> 
> try:
> 
> ```
> cat /proc/bus/input/devices
> sdljoytest -skip_loop
> cat /emuelec/configs/advmame/advmame.rc
> ```
> 
> copy/paste all output in gist, and provide link.
> 
> Also in the gamelist if you press SELECT/BACK >> Advanced Game Settings >> Emulator/Core, you can change to a different one if need be and check if they work.

ok @Langerz82 heres the Gist

<script src="https://gist.github.com/Certifr1ed/23cf527cba28d7d6e9044a334fef168b.js"></script>

Certifr1ed, 2024-10-08:
@Langerz82 https://gist.github.com/Certifr1ed/23cf527cba28d7d6e9044a334fef168b

Langerz82, 2024-10-08:
Please copy/paste one line at a time and make sure they run properly. only 1 line ran ok in your output.

Certifr1ed, 2024-10-08:
@Langerz82 got it here you go
https://gist.github.com/Certifr1ed/03a61784287031d5c971edb25a89abf9

Langerz82, 2024-10-08:
And you say the game runs but no input occurs right? I looked at the logs and can't find anything strange with it. Maybe the name of the input device in the advanced mame config files might be incorrect because it replaces the "/" with underscores "_". But other than that it doesnt look suspicious.

Langerz82, 2024-10-08:
@Certifr1ed 
Ok got a good idea I need output of the following command too:
```
advj
```
I should of thought to use this earlier.

Certifr1ed, 2024-10-08:
@Langerz82 yes game runs perfectly in mame folder but controller does nothing have to unplug device to restart.

adjv
https://gist.github.com/Certifr1ed/f0c959fbf805c2e47ef41ba8b0642cf2

FOUND A FIX thank you so much for your help tho! your awesome
the fix ended up being switching the emulator to mame2010 for that specific game and that allowed controller to fully work and still access retroarch.

So i suppose when i run into an arcade game that gives red error screen i will move that specific rom to mame folder and then switch from advancemame or auto on emulator settings to mame 2010.

it appears the auto feature doesn't select the best emulator right away in arcade specific games, wish they could improve that in future versions. Anyways thanks for your time have a great week!

Langerz82, 2024-10-08:
It's great you have an alternative fix. I can see the issue that is causing no input in Advmame. In advj the controllers listed name is 'shanwan_pcps3android' but the config lists it as 'shanwan_pc_ps3_android'. So I would just need to strip the '/' instead of replacing them with underscores.

run these commands one at a time to get your Advmame controllers working:
```
cd /emuelec/bin
wget https://raw.githubusercontent.com/Langerz82/EmuELEC/refs/heads/dev_advmame_joy_name_fix/packages/sx05re/emulators/advancemame/bin/set_advmame_joy.sh
chmod +x /emuelec/bin/set_advmame_joy.sh
```
And if all is correct, it should now use the corrected auto joypad file.

Langerz82, 2024-10-08:
Thanks for you report and your patience, future versions should include the fix for you.

Certifr1ed, 2024-10-08:
Thanks again I run command in ssh? The one found here is updated?

Langerz82, 2024-10-09:
yep in SSH run these commands one line at a time do:
```
cd /emuelec/bin
wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/be03a664c59978ea526086f586c7d66b1350e29e/packages/sx05re/emulators/advancemame/bin/set_advmame_joy.sh
chmod +x /emuelec/bin/set_advmame_joy.sh
```

Certifr1ed, 2024-10-10:
@Langerz82 that worked thank you for your hardwork!

Langerz82, 2024-10-11:
The previous code had a bit of an ordering bug, If you want the latest script which was corrected do the following:
```
rm /emuelec/bin/set_advmame_joy.sh
cd /emuelec/bin
wget https://raw.githubusercontent.com/EmuELEC/EmuELEC/ce58b1ca73ee2a2ee279d6b51e09878e25bf0178/packages/sx05re/emulators/advancemame/bin/set_advmame_joy.sh
chmod +x /emuelec/bin/set_advmame_joy.sh
```
And that should use the latest gamepad config file.

If you want to remove the overided file for any reason just do:
```
rm /emuelec/bin/set_advmame_joy.sh
```
I will close this issue, but if you have any issues feel free to message on here and I will re-open it.

closed 2024-10-11 by Langerz82