EmuELEC/EmuELEC #1429: PS3 type controller cannot exit games on ppsspp standalone emulator in Emuelec 4.8 TEST

alonsocr23, 2024-10-17:
Hi, I have been trying to try to exit any psp game that I emulate with the ppsspp standalone emulator but I have tried setting the hotkey button to any other button but nothing, because I know that the problem comes from the button assigned as hotkey. I have tried with a PS4 type controller and with that one it works. On the other hand with the emulator Libretro: ppsspp if it works to exit the game the PS3 controller.

 - EmuELEC Version: 4.8 TEST
 - SOC: [S905x3]
 - Brand: [HK1 Box]

Langerz82, 2024-10-17:
Can you login to your device with your PS3 controller connected and type in:
```
sdljoytest -skip_loop
```

alonsocr23, 2024-10-17:
Hi Langerz82 I have never done that SSH thing, I can try but I need you to explain step by step how to do it, that is new to me.

Langerz82, 2024-10-17:
Just wondering did you manually map the PS3 Controller Buttons in (START >> Bluetooth and Controller Settings, Map Controller), and when you get to the hotkey you can re-assign a button such as the SELECT/BACK key.

alonsocr23, 2024-10-18:
> ¿Puedes iniciar sesión en tu dispositivo con SSH con tu controlador PS3 conectado y escribir:
> 
> ```
> sdljoytest -skip_loop
> ```
> 
> Ah, y pega el resultado en un archivo de texto y cárgalo aquí.

[image]

Hi Langerz, does it work like this? it's a screenshot. Another thing for version 4.8 you must add in the file es_systems.cfg the extension .CHD for PSP and PSPMINIS because it is not there and I had to add it myself.

Langerz82, 2024-10-18:
The log shows you have the guide button assigned as value b161 which I dont think is a valid button number. You need to manually assign the hotkey button as back (select button) which would be b10, manually mapping the button assignments in ES.

alonsocr23, 2024-10-18:
> The log shows you have the guide button assigned as value b161 which I dont think is a valid button number. You need to manually assign the hotkey button as back (select button) which would be b10, manually mapping the button assignments in ES.

The question is which button should I select as hotkey other than the SELECT button, since there are no more free buttons and I also tried other buttons like L2 or L3 but they didn't work either.

alonsocr23, 2024-10-18:
> It doesn't need to be a free button, you can use the select button again.

but I have already tried it and it does not work, in the screenshot I sent you from Putty the hotkey function is assigned to the SELECT button.

Langerz82, 2024-10-18:
You might have to manually adjust the guide button value in SSH, copy paste one line at a time:
```
cd ~/.config/SDL-GameControllerDB
cat gamecontrollerdb.txt | grep 0500a5d049190000020400001b010000 | sed -i "s|guide:b161|guide:b10|" gamecontrollerdb.txt
```

alonsocr23, 2024-10-18:
> You might have to manually adjust the guide button value in SSH, copy paste one line at a time:
> 
> ```
> cd ~/.config/SDL-GameControllerDB
> cat gamecontrollerdb.txt | grep 0500a5d049190000020400001b010000 | sed -i "s|guide:b161|guide:b10|" gamecontrollerdb.txt
> ```

just like that?
[image]

Langerz82, 2024-10-18:
no 1 line at a time and press enter for each command.

worstcase-scenario, 2024-10-18:
Don't forget that there are also gamecontrollerdb.txt files for ppsspp under
 
./storage/.config/ppsspp/assets/gamecontrollerdb.txt and
 ./storage/roms/bios/PPSSPP/gamecontrollerdb.txt
 
 Perhaps it's better to delete them first....

alonsocr23, 2024-10-18:
> @alonsocr23 no 1 line at a time and press enter for each command.

doing it as I have shown you it works although the ppsspp standalone menu is not shown but pressing SELECT+START does exit the game. If I reconfigure the controller in ES will it be deconfigured?

Langerz82, 2024-10-18:
@worstcase-scenario the auto gamepad would copy over that.

@alonsocr23 Yes it will copy over the record again, to access the game menu in PPSSPPSDL you might need to use a keyboard.

alonsocr23, 2024-10-18:
So do I leave it like this or do I do something else?

Langerz82, 2024-10-18:
Shit I messed up the command, it would have changed all the guide buttons for all game controllers. You will need to restore gamecontrollerdb.txt. Run the following:
```
cp /usr/config/SDL-GameControllerDB/gamecontrollerdb.txt ~/.config/SDL-GameControllerDB
```
I will send you a command on how to override the record a bit later.

alonsocr23, 2024-10-18:
so what you sent me and that seems to have solved the problem is wrong? well, I hope you tell me, thanks.

Langerz82, 2024-10-18:
First make sure you have restored the gamecontrollerdb file with:
```
cp /usr/config/SDL-GameControllerDB/gamecontrollerdb.txt ~/.config/SDL-GameControllerDB
```
If you do :
```
sdljoytest -skip_loop
```
where it says mapping, you will have the controls and there associated values. If you want to change the guide button (which is considered the hotkey button) to say the back/select button it would be b10 for example. Then in the following commands you can change the guide button by replacing the value, however it does not need to b10 it can be left thumbstick for example with is b13
```
cd ~/.config/SDL-GameControllerDB
GCDBLINE=$(cat gamecontrollerdb.txt | grep -n 0500a5d049190000020400001b010000 | cut -d':' -f1 )
sed -i "${GCDBLINE}s/guide:b[0-9]*/guide:b10/" gamecontrollerdb.txt
```
Commands explained:
1st line gets into the gamecontrollerdb.txt directory.
2nd returns the line number to replace the text in the following line (searching by game controller guid).
The 3rd line is where you can change "guide:b10" for instance to "guide:b13" if you want.

Hope that clears things up, And apologies for my previous command stuffing up your gamecontrollerdb.txt file before.

Langerz82, 2024-10-18:
Just an update my previous one-line command was also stuffed.

alonsocr23, 2024-10-18:
> sed -i "${GCDBLINE}s/guide:b[0-9]*/guide:b10/" gamecontrollerdb.txt

now I am testing but the START button does not work and I can't move to the next screen in a PSP game.

Langerz82, 2024-10-18:
Remap your controller in ES so the record update new. Then try and apply the commands.

alonsocr23, 2024-10-18:
> Reasigna tu controlador en ES para que se actualice el registro. Luego, intenta aplicar los comandos.

Sorry, this last problem was my fault. But now another thing I would like to know is how to save the game if pressing only the SELECT button does not show the PPSSPP Standalone menu.

Langerz82, 2024-10-18:
That's the trade off, you will probably need a keyboard to access the game menu.

alonsocr23, 2024-10-18:
> Esa es la desventaja: probablemente necesitarás un teclado para acceder al menú del juego.

but for example with a PS4 type controller if the menu comes out with the button between the two joysticks, I could try to set the button between the two joysticks on the PS3 controller as HOTKEY but I do not know what number is equivalent, could you help me on that?

Langerz82, 2024-10-18:
if it does not register as a button in the ES Controller Mapping then it is not possible.

alonsocr23, 2024-10-18:
> Si no se registra como un botón en el mapeo del controlador ES, entonces no es posible.

if it is assigned as a button in ES specifically button 25

Langerz82, 2024-10-18:
If it works in ES but not in PPSSPP-SDL you will have to make a feature request here:
https://github.com/hrydgard/ppsspp/issues

alonsocr23, 2024-10-18:
> Si funciona en ES pero no en PPSSPP-SDL, tendrás que realizar una solicitud de función aquí: https://github.com/hrydgard/ppsspp/issues

ok, let's see if we are lucky, thanks

Langerz82, 2024-10-18:
Actually it would be gptokeyb that is responsible for exiting, I'm not sure it will accept button26 or not the code is here:
https://github.com/EmuELEC/gptokeyb/issues

Langerz82, 2024-10-18:
A cleaner 1 one line command would be:
```
sed -i '/(GUID)/s/guide:b[0-9]*/guide:b(BUTTONCODE)/' ~/.config/SDL-GameControllerDB/gamecontrollerdb.txt
```
so in your case for guid to set to back/select it would be:
```
sed -i.bak '/0500a5d049190000020400001b010000/s/guide:b[0-9]*/guide:b10/' ~/.config/SDL-GameControllerDB/gamecontrollerdb.txt
```

alonsocr23, 2024-10-18:
> Primero asegúrese de haber restaurado el archivo gamecontrollerdb con:
> 
> ```
> cp /usr/config/SDL-GameControllerDB/gamecontrollerdb.txt ~/.config/SDL-GameControllerDB
> ```
> 
> Si lo haces:
> 
> ```
> sdljoytest -skip_loop
> ```
> 
> Donde dice mapeo, tendrás los controles y sus valores asociados. Si quieres cambiar el botón guía (que se considera el botón de acceso rápido) para que sea el botón de atrás/seleccionar, sería b10, por ejemplo. Luego, en los siguientes comandos, puedes cambiar el botón guía reemplazando el valor, sin embargo, no es necesario que sea b10, puede ser el joystick izquierdo, por ejemplo, que es b13.
> 
> ```
> cd ~/.config/SDL-GameControllerDB
> GCDBLINE=$(cat gamecontrollerdb.txt | grep -n 0500a5d049190000020400001b010000 | cut -d':' -f1 )
> sed -i "${GCDBLINE}s/guide:b[0-9]*/guide:b10/" gamecontrollerdb.txt
> ```
> 
> Explicación de los comandos: la primera línea ingresa al directorio gamecontrollerdb.txt. La segunda línea devuelve el número de línea para reemplazar el texto en la siguiente línea (buscando por el GUID del controlador de juego). La tercera línea es donde puedes cambiar "guide:b10", por ejemplo, a "guide:b13" si lo deseas.
> 
> Espero que esto aclare las cosas y pido disculpas por mi comando anterior que estropeó tu archivo gamecontrollerdb.txt.

Hey langerz, I'm back, I've finally managed to do what I was telling you. Thanks to this answer that you put here. Let me explain:

I followed the steps that you put me in which you assign the button 10 (the SELECT) as HOTKEY but there was not the ppsspp standalone menu, then as I have seen that when I configured the PS3 controller type if assigned as HOTKEY the button between the two joysticks assigned me the button 25, so as in the line that you assigned me b10 I changed the b10 by the b25 and BOOOMMM surprise worked. Now when I press button 25 the ppsspp standalone menu appears and if I press b25+start the game exits. So all that, perfect, you just need to do it for version 4.8 of emuelec. If you need help checking 4.8 with that problem solved, do not hesitate to ask me for help and I will be happy to check. Thank you very much for your patience and help.

Langerz82, 2024-10-18:
@alonsocr23 
Good to see it works for you. 

The scripts that are responsible for generating the input configs are here:
https://github.com/EmuELEC/EmuELEC/tree/dev/packages/sx05re/emuelec-emulationstation/config/scripts/configscripts

I'll see what I can do about enabling guide buttons that have very high values (in your case it was 25).

alonsocr23, 2024-10-20:
> @alonsocr23 Good to see it works for you.
>
> The scripts that are responsible for generating the input configs are here: [...]

hello @Langerz82  by changing what I did now in the Arcade games for example I can not put credits, so I stay the same, let's see if before we release version 4.8 of emuelec we can solve it. 
Now I can't get the PS3 controller to work in fbneo to enter the credits.