EmuELEC/EmuELEC #1432: set individual game borders

Ernando79, 2024-11-05:
Hello, I'm not able to configure the game individually in ee_emu.framebuffer. I go to the game I want to configure, press and hold the button, go to the game option and there I configure the borders, but it doesn't work. Is it possible to configure the borders individually for each game? I'm using emuelec 4.8 test.

Langerz82, 2024-11-05:
To test:
https://github.com/Langerz82/EmuELEC/releases/tag/v4.8TEST20241106

Langerz82, 2024-11-06:
Ok here is a new release:
https://github.com/Langerz82/EmuELEC/releases/tag/v4.8TEST20241106123254

See PR:
https://github.com/EmuELEC/EmuELEC/pull/1433

NOTE - framebuffer dimensions setting is no longer tied to the display resolution, that means if you change the framebuffer resolution for all emu's (in emuelec settings >> danger zone) it will always change the frambuffer to it setting regardless of the resolution you have in-game even when it's set to auto or you have set the native video differently. If you don't want the frame buffer to set differently keep it at auto.
Border changes should work fine, however they are no longer tied to resolution either. So if you say run native video 640x480 (640x480p60hz) and set a border value, If you change the native video to a new resolution say 1024x768 the border will apply to that resolution as well. To remove bordering you need to set the values, to top left right bottom all 0.

Langerz82, 2024-11-11:
This PR will fix the users issue:
https://github.com/EmuELEC/EmuELEC/pull/1436

Closing.

closed 2024-11-11 by Langerz82