EmuELEC/EmuELEC #1434: Wiimote (button) input is not recognized because of lircd-process

worstcase-scenario, 2024-11-07:
**Describe the bug**
After succesfully connecting an original Wiimote via bluetooth, the controller cannot be configured via the menu, as no pressed buttons are being recognized.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Wiimote via Bluetoothctl (right now pairing the controller via the menu does not work)
2. Try to map the buttons - it won't work, no input is received.

**Expected behavior**
All the buttons should be recognized by EmuElec and the Wiimote should work as a controller.

**EmuELEC version:**
 - Version v4.8-TEST-04212024

**Device (please complete the following information):**
 - Amlogic s905x2
 
SOLUTION:

The reason of this issue lies in the running lircd-process. If this process is killed, the Wiimote-buttons are instantly recognized.

Workaround:
1. Either disable the process completely or 
2. Use the following script in custom_start.sh:

#!/bin/bash

# The process name to search for
PROCESS_NAME="/usr/sbin/eventlircd -f --evmap=/etc/eventlircd.d --socket=/run/lirc/lircd"

# Search for the process and terminate it if it's running
PID=$(pgrep -f "$PROCESS_NAME")
if [ -n "$PID" ]; then
    echo "Process found with PID: $PID. Terminating the process..."
    kill $PID
    echo "Process $PID has been terminated."
else
    echo "Process '$PROCESS_NAME' not found."
fi

skiiingray, 2024-11-13:
I can't look for my BT mac even controller info via "bluetoothctl list" when i enabled BT in emuelec setting of S905x4 device that is featured BT 4 inbox and working in android. How can I activate BT first?

worstcase-scenario, 2024-11-13:
Well, does Bluetooth work in emuelec? Can you use Bluetooth controllers?

skiiingray, 2024-11-13:
weirdly i got the error message by "No default controller available" when commanding **power o**n, **_discoverable on** and the later. Neither command_by_command in ssh nor editing to run file of /storage/.config/custom_start.sh it did work. No idea what's the issue in emuelec OS environment as I can utilize BT in default to pair BT devices like earphone and gamepad in nand android OS.

worstcase-scenario, 2025-06-17:
Issue will be fixed with https://github.com/EmuELEC/EmuELEC/pull/1500