EmuELEC/EmuELEC #1437: upgrade RetroArch

2494654220, 2024-11-13:
hello,I am using emuelec version 3.3. How can I upgrade RetroArch without upgrading emuelec

Langerz82, 2024-11-13:
update the hash value, and recompile the build.

https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/libretro/retroarch/package.mk

Also this is not a issue, just a general question please use the discord.

closed 2024-11-13 by Langerz82

2494654220, 2024-11-13:
> update the hash value, and recompile the build.
> 
> https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/libretro/retroarch/package.mk
> 
> Also this is not a issue, just a general question please use the discord.

After the compilation and build process, I receive a directory named "retroarch-1e1b24c35996436e6cf0fe50ec046e2e03a0dbfa". I'm uncertain if this is correct. Should I replace the entire directory on the SD card with this one?

Langerz82, 2024-11-14:
First you need to be in the 3.9 emuelec tag branch.
you need to build the whole project and you should get the output files in the target directory. Then you copy that onto an sd card.

Langerz82, 2024-11-16:
Hmm it seems the patches are failing. Try removing the files from the retroarch directory in master_32bit, and copying the retroarch directory from the latest dev version. Make sure your master 32bit version is v3.9.

2494654220, 2024-11-16:
> Hmm it seems the patches are failing. Try removing the files from the retroarch directory in master_32bit, and copying the retroarch directory from the latest dev version. Make sure your master 32bit version is v3.9.

I updated the directory with files from the dev branch (Figure 1)
[image]

 after recompiling, some patches were prompted to be skipped, followed by an error message (Figure 2).
[image]

Langerz82, 2024-11-16:
did you also update the hash value to the dev version?

2494654220, 2024-11-16:
> did you also update the hash value to the dev version?

Yes, the package.mk file is also from the DEV branch.

Langerz82, 2024-11-16:
hmm not sure maybe you have to update some of its dependencies as well.

2494654220, 2024-11-16:
> hmm not sure maybe you have to update some of its dependencies as well.

Oh, I may not be able to solve it. Thank you very much for your guidance.