EmuELEC/EmuELEC #1442: SOLARUS controller mapping for Amlogic-devices

worstcase-scenario, 2024-11-22:
**Describe the bug**
When trying to play a solarus game on Amlogic devices, EmuElec seems to use the pads.ini-file, which has entries in it that mostly don't match the buttons in es_input.cfg.

**To Reproduce**
Start up a solarus game and try to use a controller - it won't work or only partly.

**Expected behavior**
Using the controller without problems.

**EmuELEC version:**
 - Version v4.8-TEST-04212024

**Device (please complete the following information):**
 - SOC: S905x2

**SOLUTION:**
Adding "Amlogic-ng" as a device in the code of solarus.sh makes EmuElec use the gptokeyb-config, which solves the problem of not-rightly configured buttons...

if [ "${EE_DEVICE}" == "OdroidGoAdvance" ] || [ "${EE_DEVICE}" == "Amlogic-ng" ]|| [ "${EE_DEVICE}" == "GameForce" ]; then 
    gptokeyb -c /emuelec/configs/gptokeyb/solarus-run.gptk &
    SDL_GAMECONTROLLERCONFIG_FILE="" solarus-run -fullscreen=yes ${1}
    killall gptokeyb

shantigilbert, 2024-12-04:
Thank you!. Can you open a PR with this change?

worstcase-scenario, 2024-12-04:
Okay, pull request started, hope I did everything right, it's my first pull request ever...

Langerz82, 2024-12-15:
Merged Fix:
https://github.com/EmuELEC/EmuELEC/pull/1454

closed 2024-12-15 by Langerz82