EmuELEC/EmuELEC #1451: How to modify the orientation of the ES system game interface？

johy-525, 2024-12-04:
The ES 4.7  UI direction is normal, but it rotates 90 degrees when entering each game interface. Where can I modify this?

thanks！

Langerz82, 2024-12-06:
Do you want ES to be rotated as well?

johy-525, 2024-12-08:
I have already implemented the rotation of the ES UI, but the game still rotates 90 degrees. How to change the selection direction of the game？ Thanks!

Langerz82, 2024-12-08:
you can only change the screen rotation for retroarch, the standalone emulators don't support rotation. It is video_rotation in retroarch.cfg, 1 = 90deg, 2 = 180deg etc. It also has the value screen_roation but I don't think the value works in EmuELEC from memory.

johy-525, 2024-12-09:
hello,I changed video_rotation=1 and the game direction did not change,
The ES UI direction refers to the successful modification of screen_orientation。

Langerz82, 2024-12-15:
try screen_rotation in retroarch.fg as well.

Langerz82, 2025-07-09:
Latest fix in dev version includes option to pass in ES arguments. EmuelecSettings >> Danger Zone >> ES Arguments.

For example you can add:
```
--screenrotate 1 --screensize 1080 1920
```
To rotate the view by 90deg.