EmuELEC/EmuELEC #1456: Fan Control Not Activating Based on Temperature on Khadas VIM3 with EmuELEC 4.7

R0dznCL, 2024-12-11:
Device: Khadas VIM3

Version: EmuELEC 4.7

Issue Type: Bug - Fan control not activating based on temperature

Steps to Reproduce:

Install EmuELEC 4.7 on Khadas VIM3 hardware.
Configure the system and run games or applications that increase the device temperature.
Observe that the fan does not activate automatically, regardless of the temperature.
Expected Behavior: The fan should activate automatically when the device reaches a certain temperature threshold, ensuring proper cooling of the hardware.

Actual Behavior: The fan remains inactive even when the device gets significantly warm, potentially leading to overheating and performance issues.

Hardware Information:

Device Model: Khadas VIM3
CPU: Amlogic A311D (4x Cortex-A73 @ 2.2GHz + 2x Cortex-A53 @ 1.8GHz)
GPU: Mali-G52 MP4
Cooling System: Integrated fan connected to the GPIO pins with PWM support.

Additional Context:

The issue occurs consistently and has been tested across multiple gaming sessions.
The fan works correctly on other operating systems such as Ubuntu or Android on the same hardware, indicating that the issue might be related to EmuELEC’s fan control configuration.

Reproducibility: This issue is consistently reproducible on the Khadas VIM3 with EmuELEC 4.7.

shantigilbert, 2024-12-11:
Its been a long time since I used the VIM3, but have you done any testing to make sure the temperature is pass the fan trigger? which if I remember correctly it was 70c

R0dznCL, 2025-02-02:
Sorry for the delayed response, but I can assure you that the temperature far exceeds 70°C. In fact, the device locks up due to the temperature increase.

I can also confirm that this is not a connection issue because when I start the device with the pre-installed native Android, the fan operates without any problems at its three speeds.

There is a different issue here. Hopefully, it can be reviewed, or you can let me know what information you need to correct this behavior.