EmuELEC/EmuELEC #1457: FAILURE: scripts/install multiemu:target has failed!

986046679, 2024-12-11:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
When I tried to compile, I got a 404 error on the webpage. I checked the link and found that the source of the link had expired.
[image]

**To Reproduce**
Steps to reproduce the behavior:
Direct compilation

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**EmuELEC version:**
 - Version 4.7
 - The latest version

**Device (please complete the following information):**
ubuntu20.4

**Additional context**
Add any other context about the problem here.

worstcase-scenario, 2024-12-11:
Read the rules first before posting:
PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

This is not a problem with Emuelec, please close this issue and use the discord for help.

986046679, 2024-12-11:
ok

closed 2024-12-11 by 986046679

986046679, 2024-12-11:
> try deleting that package file in your sources folder and in your build directory and re-trying.

I've tried, but the same error keeps appearing. I tried accessing the source website directly, but I got a 404 error.

Langerz82, 2024-12-11:
Try downloading the file directly and do a sha256sum on it and see if you get the same hash. If you do and you are sure it's incorrect then update the package file with the hash you are getting.

986046679, 2024-12-11:
> Try downloading the file directly and do a sha256sum on it and see if you get the same hash. If you do and you are sure it's incorrect then update the package file with the hash you are getting.

I tried it and it worked, but I still have a question in version 4.3 if you have a fix for it
[image]

Langerz82, 2024-12-11:
Not sure about that error. Check with package owner for bug and fix for version maybe?
@shantigilbert any idea what he can do?

Just a question, why are you compiling an older version 4.3?

986046679, 2024-12-11:
> Not sure about that error. Check with package owner for bug and fix for version maybe? @shantigilbert any idea what he can do?
> 
> Just a question, why are you compiling an older version 4.3?
> 
> checked error online: you might need to update golang package to version 1.17 or higher.
> 
> try copy these files into your 4.3 version, and see if it works. you might need an earlier version. https://github.com/EmuELEC/EmuELEC/tree/dev/packages/addons/addon-depends/go https://github.com/EmuELEC/EmuELEC/commits/dev/packages/addons/addon-depends/go/package.mk

Since I want to run this version on the s905L, I will try the method you mentioned

986046679, 2024-12-16:
> Not sure about that error. Check with package owner for bug and fix for version maybe? @shantigilbert any idea what he can do?
>
> Just a question, why are you compiling an older version 4.3?
>
> checked [...]

Unfortunately, when I tried to follow your method, I got another error.
[image]