EmuELEC/EmuELEC #1461: Version 4.8 TEST time is displayed incorrectly in my timezone

joelsaacks, 2024-12-29:
**Describe the bug**
I've got the time set to Africa/Johannesberg in the menu options. The time seems to be shown as UTC
I tried a few others. It shows the same time.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on time display
2. Restart
3. Time at the bottom is two hours behind.

**Expected behavior**
Correct time

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version 4.8-TEST

**Device
 - SOC: S095x3
 - Brand: X96
 - Model AIR 4GB/100 Mbit
 - Version:

shantigilbert, 2024-12-29:
Are you connected to the internet?

joelsaacks, 2024-12-29:
Yes. 24x7. Time is set as soon as I boot. I'm even scraping at the moment. So definitely so.

joelsaacks, 2024-12-29:
Time is showing as 18:31 even though its 20:31

Langerz82, 2025-06-17:
Africa/Johannesberg is GMT+2 so it looks like it's using the GMT and not applying the timezone change.

joelsaacks, 2025-06-17:
> Africa/Johannesberg is GMT+2 so it looks like it's using the GMT and not applying the timezone change.

Exactly that.