EmuELEC/EmuELEC #1470: How to adapt my RK3568 development board?

LeoD218, 2025-02-12:
I designed a development board based on the RK3568 and flashed the EmuELEC-RK356x.aarch64-4.7-rk356x.img.gz image. I found that when I connected HDMI, there was audio output but no video output, which should be due to the device tree not being configured correctly. How can I compile the Linux kernel separately, modify the device tree, and then package the image?

shantigilbert, 2025-02-21:
Can you join Discord? this seems something better suited to discuss there. 

If not, you can follow this WiKi entry: 

https://github.com/EmuELEC/EmuELEC/wiki/DEV:-How-to-compile-toolchain-only

That will help you compile the toolchain, after that you can compile any package you need (e.g. kernel)

(you will need to check that the project is correct as I don't recall at the moment)
`PROJECT=RK356x ARCH=aarch64 DISTRO=EmuELEC ./scripts/build kernel` 

Basically, the RK356x image we released was made for 2 specific boards, each have their own u-boot, but AFAIR they share the same kernel. I am just assuming here, but if you have sound but no video it is either u-boot that is not configured properly (using the custom versions for those 2 boards) or as you say the device tree needs modification.

LeoD218, 2025-02-24:
Thanks for the reply, I'll give it a try！

closed 2025-02-24 by LeoD218