EmuELEC/EmuELEC #1475: Discord invitation link is not valid (expired?)

soadzoor, 2025-03-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [EmuElec's GitHub Page](https://github.com/EmuELEC/EmuELEC)
2. Click on the [discord link](https://discord.gg/jQWCFwTn5T)
3. See error:

[image]

**Expected behavior**
I expect to be able to join the discord channel.

**Screenshots**

[image]

worstcase-scenario, 2025-03-19:
Try this: https://discord.gg/sEKkTsDe

soadzoor, 2025-03-19:
> It's working for me... But you can try this (as a quick solution): https://discord.gg/sEKkTsDe

this one worked for me, thanks!

closed 2025-03-23 by Langerz82