EmuELEC/EmuELEC #1482: Mapping problem for player 1 and player 2 controls

Rocketgirl2021, 2025-03-30:
I would like to leave an important bug in the button mapping for 1 player and 2 players. In versions emuelec 4.7 and 4.8 beta there is this bug, in version 4.6 this problem does not exist, everything works correctly, could it be fixed?

Langerz82, 2025-04-05:
What is the issue exactly does the controller not get picked up as a valid controller.
Is it bluetooth and does it not connect automatically?
Can you ssh in and type: 
```
sdljoytest --skip_loop
```
and post all the output?
details on how to ssh in are here:
https://github.com/EmuELEC/EmuELEC/wiki/How-to-SSH-into-EmuELEC

Langerz82, 2025-06-16:
@Rocketgirl2021 Can you elaborate on what was the issue exactly?

closed 2025-06-18 by Langerz82