EmuELEC/EmuELEC #1483: Odroid C2 guide

soadzoor, 2025-03-31:
Hey folks,

I have an odroid c2. Does anyone have a step-by-step guide on how to fire up emuelec on that? I already flashed the generic (non-ng) version of emuelec 4.3 on a micro sd card with balenaetcher, pasted "gxbb_p200_2g_odroidc2.dtb" to the root of the sdcard and renamed it to "dtb.img".

Now, it doesn't boot up (it shows a black screen only), and the guide says something about needing to hold the "reset" button, although I can't really see any reset buttons on the odroid c2, no 3.5mm jack, either.

Then it says "Alternatively if your device has no reset button, you can boot into android and use an app called "Reboot to CoreELEC" (Or "Reboot to LibreELEC") while the SD is inserted", but how could I boot into android, if the sdcard I'm inserting already has "emuelec" on it?.. I'm soooo confused... any help is appreciated!

_The guide is not very helpful, it only mentions "p200/p201" for "s905" chip, and there are tons of dtb files with "p200" and "p201" in their names... I think the one with "odroidc2" in its name should be the right one, although if I don't read through all the files, it's not very obvious (based on the guide) that that's the one I need.
The odroid c2, while being one of the few particular devices that are mentioned as "supported", turns out that it's not even supported since version 4.3, so it's really difficult to get a "proper" device for emuelec, it seems... Anyway..._

worstcase-scenario, 2025-06-11:
It's simple: you eject the SD card, then boot into android, insert the SD card again and reboot the device by using the app.

soadzoor, 2025-06-16:
> [@soadzoor](https://github.com/soadzoor) let us know if the reply works for you. If it doesn't reply here and I will re-open the issue.
@Langerz82  thank you for your time!

> It's simple: you eject the SD card, then boot into android, insert the SD card again and reboot the device by using the app.
@worstcase-scenario 
Thank you for willing to help! Although, in case of an odroid c2, the steps of _"eject the SD card, then boot into android"_ doesn't really make any sense. An odroid c2 can only boot any OS from an sdcard (or emmc module) as far as I know, it doesn't have an internal memory, and it doesn't have android on it, by default, either.

I ended up downloading the "odroid c2" version of emuelec, specifically this version: [EmuELEC-Amlogic.aarch64-4.3-Odroid_C2.img.gz](https://github.com/EmuELEC/EmuELEC/releases/download/v4.3/EmuELEC-Amlogic.aarch64-4.3-Odroid_C2.img.gz), because 4.3 is the latest version that supports odroid C2.

The wiki page, and other documents didn't really help, because based on those, I was supposed to download the `generic` version, and then replace the proper dtb file. It's even more confusing than one might think, because a dtb file labeled as `odroid c2` is present in the general img file, if you install it on an sd card.

Anyway, I figured it out, but it would've been a lot faster, if it was documented properly on the "supported devices" page, or the "how to install" wiki.

Also, almost all of the linked devices in the examples of "supported devices" are not available, the links are outdated, etc. It would be a good idea to refresh those links, I believe.

Langerz82, 2025-06-17:
I updated the wiki to reflect the different steps on installing Odroid C2, among some others.

See:
https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install