EmuELEC/EmuELEC #1489: Odyssey 2 wrong roms path

sanviler, 2025-05-04:
**Describe the bug**
According to es_systems.cfg the Odyssey 2 roms path folder should be "/storage/roms/odyssey2", as it will should system name, platform and theme, instead of "/storage/roms/odyssey".

Having this discrepancy makes bezels inside the "/bezels/odyssey" directory not being read, because ES is telling RA to look for them in a directory named "odyssey2".

Users commonly know that they have to name the bezels directories after the roms directories, which in this case will lead in RA not showing up any bezel placed inside the "odyssey" directory.

**EmuELEC version:**
 - Version 4.8

**Additional context**
The solution would be to rename the "/storage/roms/odyssey" directory to "/Storage/roms/odyssey2" and contextually changing the "<path>" tag in the es_systems.cfg as following:

`<system>
		<name>odyssey2</name>
		<fullname>Magnavox Odyssey 2</fullname>
		<manufacturer>Magnavox - Philips</manufacturer>
		<release>1979</release>
		<hardware>console</hardware>
		<path>/storage/roms/odyssey2</path>
		<extension>.bin .BIN .zip .ZIP .7z .7Z</extension>
		<command>emuelecRunEmu.sh %ROM% -P%SYSTEM% --core=%CORE% --emulator=%EMULATOR% --controllers="%CONTROLLERSCONFIG%"</command>
		<platform>odyssey2</platform>
		<theme>odyssey2</theme>
		<emulators>
			<emulator name="libretro">
				<cores>
					<core default="true">o2em</core>
				</cores>
			</emulator>
		</emulators>
	</system>`

cheers!

shantigilbert, 2025-06-10:
Roms partition needs to support symlinks, fat32 does not. If that is the correct rom path the we should just update it and let users know of the change. I don't believe symlinks would be a good idea on a rom path.

Langerz82, 2025-06-18:
Will be fixed as of patch:
https://github.com/EmuELEC/EmuELEC/pull/1503

So closing issue.

closed 2025-06-18 by Langerz82