EmuELEC/EmuELEC #1509: Portmaster games using Westonpack Runtime doesn't work

25minutes, 2025-06-30:
**Describe the bug**
Portmaster games using Westonpack Runtime doesn't work. I've tested a few and it's just black screen. For example, Steamworld Dig (https://portmaster.games/detail.html?name=steamworlddig). Is it possible to make it work on EmuELEC?

**To Reproduce**
Steps to reproduce the behavior:
Just install the game via portmaster, and import the game data if required.

**Expected behavior**
A clear and concise description of what you expected to happen.
Make Westonpack Runtime games work on EmuELEC.

**EmuELEC version:**
 - Version 4.7

**Device (please complete the following information):**
 - SOC: S905X2

worstcase-scenario, 2025-06-30:
Update to 4.8-test and also update PortMaster. 
Then copy the .sh file into the installer folder and run it. 
It is working for me (see screenshot). 
Other games that are using westonpack and I personally can confirm are also working are: Braid and SuperHexagon. 

EmuELEC is officially not supported by the PortMaster crew, so there is no guarantee that all games are going to run on EmuELEC, too. Check the list (https://github.com/EmuELEC/EmuELEC/wiki/List-of-Ports-running-on-EmuElec-(work-in-progress)) to see which games are working and which don't. If you have any problems with a port, you should ask nicely on the PortMaster discord for support and perhaps you will get help from them.

[

[image]

](url)

25minutes, 2025-06-30:
Thanks for the info. The list is very helpful. I will try again with 4.8-test and report back.

25minutes, 2025-07-01:
Yes, it works. Thank you very much.

closed 2025-07-01 by 25minutes