EmuELEC/EmuELEC #1528: Will you add Windows emulation to run applications on Emuelec?

Temixson, 2025-09-05:
The Kirin 710 runs World Of Warcraft 4.3.5a 20-25 FPS

Temixson, 2025-09-05:
https://4pda.to/forum/index.php?showtopic=992239&st=14880#entry115074205

Temixson, 2025-09-05:
Make it possible to use a "non-linear depth buffer" without this function you cannot play GTA SA with PS2 textures on Android!!!

Temixson, 2025-09-05:
TEXTURE RESOLUTION EXACTLY 210 WITHOUT THIS EMULATORS ARE NOT PLAYABLE!!!!!

Temixson, 2025-09-05:
Three processor cores are enough for games, otherwise there will be overheating.
I'm on Android playing PSP games on S905x3.

Temixson, 2025-09-05:
Gtools increased FPS a lot

worstcase-scenario, 2025-09-05:
Emuelec is not able to do Windows emulation. You can only run a windows 95 image via dosbox.

Temixson, 2025-09-07:
So you will create a separate group for Windows XP program emulation kernels?

Temixson, 2025-09-07:
https://ezwow.org/topic/149570-praktika-ispolzovaniia-emuliatora-windows-dlia-zapusk/?p=1939496

worstcase-scenario, 2025-09-07:
No. There is no kernel. As I said, the only option is to run win 95 on dosbox, and try to run games there. But the performance is quite bad.